microsoft/AirSim #631: Unreal window just closes

Kjell-K, 2017-11-23:
@sytelus  @all Since the new AirSim updates, my test environment crashes after a while, when I perform my reinforcement learning.  [Here my project](https://github.com/Kjell-K/AirGym)

- With the AirSim versions from early October v1.1.2, NEVER had problems even running 500k steps
- I do packaging and run.bat in windowed mode
- My environment is pretty small (it is based on Blocks)
- I do not get any debug message. Unreal is just closed when I check
- Sometimes after this appears after 90k steps sometimes after 2k steps
- My PC is absolutely sufficient. 16GB RAM, Core i7, GTX 1070

- Set up: VS 2015, Unreal Engine 4.16, Win10

Any suggestions? The new updates of reset etc. are amazing but this is really an issue for progressing in RL.

cangokalp, 2017-11-24:
Yeah, I'm having a similar problem. Can't perform any training as simulation often crashes randomly. Sometimes it would crash after a day sometimes after 5 hours... it seems to be random. When it crashes I usually get an access violation error too but the error differs every time. Not sure what causes it but one thing that I notice; debugger opens up locked xthread and list files when there is a crash.

cangokalp, 2017-11-24:
Here are some examples;
[image]
[image]
[image]
[image]

Kjell-K, 2017-11-24:
@cangokalp It looks similar to my issues. It get pointed to list.

For now I am using the AirSim Version from early October (where no reset is implemented) as a work around. . . Not ideal.

sytelus, 2017-11-28:
I'm looking in to this issue. First I tried to reproduce this by writing [this simple script](https://github.com/Microsoft/AirSim/blob/a2d81ff5837a10c26c9347ae27d5e0785bfde3a1/PythonClient/drone_stress_test.py). Even after running few 100s iterations, I haven't seen the crash yet. Could you please help suggest modification to this script that would generate the crash more often? 

Also, if you can get screenshot where the "Call Stack" window is visible, that would be much more useful.

Kjell-K, 2017-11-28:
@sytelus First I suggest to let the drone collide and then as result reset. Second a few 100 iterations will not be enough since the crash occurs random. As I wrote, I had the crash between 2.000 and even 120.000 steps. Therefore it is somehow random with a very small probability to occur. But for RL almost certain to occur with large iteration number. 

[Here a test script](https://github.com/Kjell-K/AirGym/blob/master/Results/drone_stress_test_kjell.py). It is strapped down from my complete project. 
Place a few feet in front of the copter home position and let him collide and then reset.

Note: Again with the most recent Version (with the dialog "Do you want to use car?"), the window just closes and there is no debug pop up.

sytelus, 2017-11-29:
Thanks for the script. I'll try this out. Meanwhile if you are running from the source code (in Visual Studio) then you might be able to see call stack.

In any case, would it be possible to share the "Saved" folder in your project. That often contains crash logs and may help to debug this. You can probably zip up that folder and share on dropbox or something...

cangokalp, 2017-11-29:
Maybe this can help, I saved it in a notepad last time 

[call_stack.txt](https://github.com/Microsoft/AirSim/files/1512207/call_stack.txt)

I'll attach another one when it crashes again.

Kjell-K, 2017-11-29:
@sytelus When I run from source code, I have other instability issues that cause a crash before our issue even occurs.

Here are my ["Saved" folders ](https://drive.google.com/open?id=1Eh2Hm317Ye32Rz4USgnPNzkjGx1L9gQ_), which contains logs and "Crashes". Once the current version of AirSim and once the version from end of October / end of November (with the fixed client.reset).

sytelus, 2017-11-30:
@Kjell-K Thanks for sending call stack before. I have guess why this might be happening and have checked-in the [potential fix](https://github.com/Microsoft/AirSim/commit/f327504ef52b261e7346ceda0316566a6a8d650f). Please try it out and let me know how it goes.

I looked at the Saved folder as well and it looks like errors before and current are same. Its possible that its the same one that above fix applies to.

Kjell-K, 2017-12-04:
@sytelus I run my RL task now for 210k steps without a crash on the potential fix. So I think you got to the root of the issue. Well done!

I will report after 500k again. .

closed 2017-12-06 by sytelus

cangokalp, 2017-12-06:
For me the issue changed and now I get this after some time randomly .  Similar to the previous issue but now instead of crashing this happens;
self.client.call('reset')
  File "...\Anaconda3\lib\site-packages\msgpackrpc\session.py", line 41, in call
    return self.send_request(method, args).get()
  File "..\Anaconda3\lib\site-packages\msgpackrpc\future.py", line 43, in get
    raise self._error
msgpackrpc.error.TimeoutError: Request timed out

Kjell-K, 2017-12-06:
@cangokalp @sytelus  +1 unfortunately. Unreal Window is still open with the quadcopter in the terminal state (collision) for reset. This happened after 125k steps and 4k successful resets. So therefore very randomly as well. 

For now, I will build the reset into a while to call another reset if home position is not reached. Possible work around.

M-Kasem, 2018-03-12:
Is there a solution to this problem?, I am using multivehicles with multi threaded client connections and I am getting random crashes sometimes after ~20k or ~40k steps, even the logger is not consistent, sometimes the logging is:
[2018.03.11-19.20.22:563][841]LogAirSim: Error: Exception occurred while updating world: reset() must be called first before update()
Some crashed logs this:
UE4Editor: ../nptl/pthread_mutex_lock.c:117: __pthread_mutex_lock: Assertion `mutex->__data.__owner == 0' failed.
Signal 6 caught.
Malloc Size=131076 LargeMemoryPoolOffset=131092 
CommonLinuxCrashHandler: Signal=6
Malloc Size=65535 LargeMemoryPoolOffset=196655

paulfauthmayer, 2018-06-27:
For me, the issue still persists. AirSim crashes with `msgpackrpc.error.TimeoutError: Request timed out` upon calling `responses = self.simGetImages([ImageRequest(0, AirSimImageType.Scene, False, False), ImageRequest(0, AirSimImageType.DepthPerspective, True, False)])`. The amount between start and crash seems random.