microsoft/AirSim #632: What is z_min in moveByManual() command

cangokalp, 2017-11-25:
```
DroneControllerBase::moveByManual(float vx_max, float vy_max, float z_min, float duration, DrivetrainType drivetrain, const YawMode& yaw_mode, CancelableBase& cancelable_action)
{
```
I want to control the drone with roll, pitch, yaw, throttle inputs - I wrote another function to do this. It's almost the same as moveByManual - except instead of RC_data, I provide the inputs.  What is z_min in this function? I need to supply z_min to the function - but not sure what it's supposed to be? Is it the z value of the starting position of the quadcopter?

sytelus, 2017-11-28:
I've added the doc string for this method here: https://github.com/Microsoft/AirSim/blob/master/PythonClient/AirSimClient.py#L496

cangokalp, 2017-11-28:
Hi, thanks for the doc string. Although, there are still some things I'm hoping to make clear.
-The function doesn't use vz_max parameter (doc string defines this parameter)
-Still not sure about z_min. This is how vz is calculated in the function; 
float vz = (rc_data.throttle / kMaxRCValue) * z_min + getZ();
if z_min is the minimum z allowed for the vehicle position as the doc string suggests then vz might reduce to getZ(). Say the starting position is 0,0,0 and the drone is on the ground. z_min would be 0 then vz would reduce to getZ() and the rc_data.thottle will be unused...

closed 2018-02-13 by sytelus