microsoft/AirSim #633: Cannot get project to recognize plugin

rohanpai, 2017-11-25:
Hi all -- any idea why I can't build the project after following instructions here: https://github.com/Microsoft/AirSim/blob/master/docs/unreal_custenv.md 

Here's the error dialogues that I see:
[image]
[image]

Thanks!

rohanpai, 2017-11-25:
The problem is that I wasn't using a C++ project -- closing

closed 2017-11-25 by rohanpai

XinGao2017, 2017-12-04:
Hello , I have encountered the same problem , how do you solve this problem , Thanks !