microsoft/AirSim #634: Px4 stil crash

AlexandreBorowczyk, 2017-11-26:
It's seem the UE4Editor crash everytime when exiting the simulation.
it doesn't seem to leave any trace in the terminal as where it fails.

sytelus, 2017-11-27:
Some more context would help...

- Which OS? 
- Are you using latest version of AirSim?
- Does this happen every time even in simplest case?
- What is the minimum steps for repro?
- Which PX4 firmware version are you using? In SITL or HITL?

AlexandreBorowczyk, 2017-11-27:
I'm sorry.
It's on ubuntu, AirSim's master, Px4 SITL.
When quitting simulation, it will crash every time.

kandai-wata, 2017-12-19:
Same thing is happening as well.
Just for sharing.
OS: Ubuntu 14.04
AirSim: master
PX4: 1.6 SITL communicating with ros for offboard control
Environment: tried both Blocks, LandscapeMountains

When I press "stop", it crashes with the error
```
[2017.12.19-06.13.45:438][320]LogBlueprintUserMessages: Early EndPlayMap Detection: Level '/Game/FlyingCPP/Maps/FlyingExampleMap.FlyingExampleMap:PersistentLevel' has LevelScriptBlueprint '/Game/FlyingCPP/Maps/FlyingExampleMap.FlyingExampleMap:PersistentLevel.FlyingExampleMap' with GeneratedClass '/Game/FlyingCPP/Maps/FlyingExampleMap.FlyingExampleMap_C' with ClassGeneratedBy '/Game/FlyingCPP/Maps/FlyingExampleMap.FlyingExampleMap:PersistentLevel.FlyingExampleMap'
[2017.12.19-06.13.45:538][320]LogBlueprintUserMessages: Late EndPlayMap Detection: Level '/Game/FlyingCPP/Maps/FlyingExampleMap.FlyingExampleMap:PersistentLevel' has LevelScriptBlueprint '/Game/FlyingCPP/Maps/FlyingExampleMap.FlyingExampleMap:PersistentLevel.FlyingExampleMap' with GeneratedClass '/Game/FlyingCPP/Maps/FlyingExampleMap.FlyingExampleMap_C' with ClassGeneratedBy '/Game/FlyingCPP/Maps/FlyingExampleMap.FlyingExampleMap:PersistentLevel.FlyingExampleMap'
[2017.12.19-06.13.45:547][320]LogALAudio: Closing ALAudio device : OpenAL Soft
```

NihalKarne, 2017-12-25:
I am also facing same problem.
I am using PX4 in SITL. Unreal crash every time I stop the simulation with this message
```
LogAirSim: Error: Exception occurred while updating world: MavLinkConnectionImpl: Error sending message on connection 'hil', details: UdpClientPort socket send failed with error: 111
LogBlueprintUserMessages: Early EndPlayMap Detection: Level '/Game/FlyingCPP/Maps/FlyingExampleMap.FlyingExampleMap:PersistentLevel' has LevelScriptBlueprint '/Game/FlyingCPP/Maps/FlyingExampleMap.FlyingExampleMap:PersistentLevel.FlyingExampleMap' with GeneratedClass '/Game/FlyingCPP/Maps/FlyingExampleMap.FlyingExampleMap_C' with ClassGeneratedBy '/Game/FlyingCPP/Maps/FlyingExampleMap.FlyingExampleMap:PersistentLevel.FlyingExampleMap'
LogBlueprintUserMessages: Late EndPlayMap Detection: Level '/Game/FlyingCPP/Maps/FlyingExampleMap.FlyingExampleMap:PersistentLevel' has LevelScriptBlueprint '/Game/FlyingCPP/Maps/FlyingExampleMap.FlyingExampleMap:PersistentLevel.FlyingExampleMap' with GeneratedClass '/Game/FlyingCPP/Maps/FlyingExampleMap.FlyingExampleMap_C' with ClassGeneratedBy '/Game/FlyingCPP/Maps/FlyingExampleMap.FlyingExampleMap:PersistentLevel.FlyingExampleMap'
LogALAudio: Closing ALAudio device : OpenAL Soft
```
it only happens with px4. simple_flight works perfectly

petergerten, 2018-11-04:
Same error message but crashing at a different time (when arming): https://github.com/Microsoft/AirSim/issues/1516

clovett, 2018-12-23:
I'm curious if [pull request 1635](https://github.com/Microsoft/AirSim/pull/1635) helps improve this situation at all...

Zarbokk, 2019-10-20:
This is still a problem because the pull request is still not merged. (1 Year ago)

zimmy87, 2021-03-10:
This issue is still repro'ing for me after px4 shutdown and receiving the following error:

LogTemp: MavLinkConnectionImpl: Error sending message on connection 'hil', details: TcpClientPort socket send failed with error: 32

OS: Ubuntu 18.04 lts
AirSim: master
PX4: 1.10.1 SITL

I also tried with #1635 applied, but this did not fix the issue, and it appeared to cause other issues as well, so we need to consider a different solution for this.

lovettchris, 2021-06-04:
I fixed a crash like this in this PR:
https://github.com/microsoft/AirSim/pull/3757

zimmy87, 2021-06-18:
This has been fixed by #3757

closed 2021-06-18 by zimmy87