microsoft/AirSim #636: Visual Studio 2017 support

brean, 2017-11-27:
I know that Visual Studio 2017 is not supported and I saw that it is commented out in the build.cmd (so someone tried it in the past). Do you know why it is not working and do you have any plans to support it in the future?
If not, what is best-practice for new developers? Am I forced to uninstall 2017 or can run 15 and 17 side-by-side?

sytelus, 2017-11-27:
I was able to build on VS2017 without any issues. Have you tried this: https://github.com/Microsoft/AirSim/issues/595#issuecomment-344139148

Are you getting any error?

brean, 2017-11-28:
No I have not, it is compiling right now/seems to work, so THANK YOU!
I just saw at https://github.com/Microsoft/AirSim/blob/master/docs/build_windows.md in the FAQ that the build with VS 2017 is not working.

brean, 2017-11-28:
When I copy the compiled file to the Blocks-folder and load that in the Unreal Editor (Current Version 4.18.1) I am getting the following error:
```
Running C:/Program Files/Epic Games/UE_4.18/Engine/Binaries/DotNET/UnrealBuildTool.exe Blocks Development Win64 -project="C:/Users/andreas/coding/AirSim/Unreal/Environments/Blocks 4.18/Blocks.uproject" -editorrecompile -progress -NoHotReloadFromIDE
Creating makefile for Blocks (no existing makefile)
Performing full C++ include scan (no include cache file)
@progress push 5%
Parsing headers for BlocksEditor
  Running UnrealHeaderTool "C:\Users\andreas\coding\AirSim\Unreal\Environments\Blocks 4.18\Blocks.uproject" "C:\Users\andreas\coding\AirSim\Unreal\Environments\Blocks 4.18\Intermediate\Build\Win64\BlocksEditor\Development\BlocksEditor.uhtmanifest" -LogCmds="loginit warning, logexit warning, logdatabase error" -Unattended -WarningsAsErrors -installed
Reflection code generated for BlocksEditor in 8,2819673 seconds
@progress pop
Performing 11 actions (4 in parallel)
[2/11] Resource ModuleVersionResource.rc.inl
[3/11] Resource PCLaunch.rc
[4/11] Resource PCLaunch.rc
SharedPCH.Engine.cpp
Blocks.cpp
Module.AirSim.gen.cpp
Module.AirSim.cpp
[8/11] Link UE4Editor-Blocks.lib
   Creating library C:\Users\andreas\coding\AirSim\Unreal\Environments\Blocks 4.18\Intermediate\Build\Win64\UE4Editor\Development\UE4Editor-Blocks.lib and object C:\Users\andreas\coding\AirSim\Unreal\Environments\Blocks 4.18\Intermediate\Build\Win64\UE4Editor\Development\UE4Editor-Blocks.exp
C:\Program Files (x86)\Windows Kits\8.1\include\um\winnt.h(534): warning C4005: 'TEXT': macro redefinition
c:\program files\epic games\ue_4.18\engine\source\runtime\core\public\HAL/Platform.h(848): note: see previous definition of 'TEXT'
C:\Users\andreas\coding\AirSim\Unreal\Environments\Blocks 4.18\Plugins\AirSim\Source\AirLib\include\vehicles/multirotor/api/DroneApi.hpp(435): warning C4265: 'msr::airlib::DroneApi::MoveOnPath': class has virtual functions, but destructor is not virtual
         instances of this class may not be destructed correctly
C:\Users\andreas\coding\AirSim\Unreal\Environments\Blocks 4.18\Plugins\AirSim\Source\AirLib\include\vehicles/multirotor/api/DroneApi.hpp(435): warning C4265: 'msr::airlib::DroneApi::MoveOnPath': class has virtual functions, but destructor is not virtual
         instances of this class may not be destructed correctly
 
You have included windows.h before MinWindows.h
All useless stuff from the windows headers won't be excluded !!!
 
c:\users\andreas\coding\airsim\unreal\environments\blocks 4.18\plugins\airsim\source\airlib\deps\mavlinkcom\include\MavLinkLog.hpp(39): warning C4265: 'mavlinkcom::MavLinkFileLog': class has virtual functions, but destructor is not virtual
         instances of this class may not be destructed correctly
C:\Users\andreas\coding\AirSim\Unreal\Environments\Blocks 4.18\Plugins\AirSim\Source\AirLib\include\vehicles/multirotor/controllers/MavLinkDroneController.hpp(172): warning C4265: 'msr::airlib::MavLinkLogViewerLog': class has virtual functions, but destructor is not virtual
         instances of this class may not be destructed correctly
[9/11] Link UE4Editor-AirSim.lib
[10/11] Link UE4Editor-AirSim.dll
   Creating library C:\Users\andreas\coding\AirSim\Unreal\Environments\Blocks 4.18\Plugins\AirSim\Intermediate\Build\Win64\UE4Editor\Development\UE4Editor-AirSim.lib and object C:\Users\andreas\coding\AirSim\Unreal\Environments\Blocks 4.18\Plugins\AirSim\Intermediate\Build\Win64\UE4Editor\Development\UE4Editor-AirSim.exp
LINK : fatal error LNK1181: cannot open input file 'ws2_32.lib'
[11/11] Link UE4Editor-Blocks.dll
LINK : fatal error LNK1181: cannot open input file 'ws2_32.lib'
ERROR: UBT ERROR: Failed to produce item: C:\Users\andreas\coding\AirSim\Unreal\Environments\Blocks 4.18\Plugins\AirSim\Binaries\Win64\UE4Editor-AirSim.dll
Total build time: 64,19 seconds (Local executor: 0,00 seconds)
```

brean, 2017-11-29:
Its working now, so I close this issue!

I followed this: https://answers.unrealengine.com/questions/597939/new-c-project-fails-to-compile-using-ue-415-and-vs.html and it is working now (reinstalling VS took some hours -.- )
Strangely enough I can only build from the normal Command Prompt (cmd.exe), when I run build in the Native VS prompt it fails because it can not find msbuild.exe.

closed 2017-11-29 by brean