microsoft/AirSim #637: C++ API simGetImages first image wrong pose

trbotha, 2017-11-28:
In the C++ API using the StereoImageGenerator example I have a problem when using simGetImages. The first image in the array is not at the correct pose and location. When reading multiple images say left front and right front, the left front which is the first in the array will be taken at a different pose than the right image. In subsequent calls the left image starts to follow the path provided but always seems one image behind. One can work around it by calling simGetImages twice, or by having the first requested image a dummy image as this only seems to affect the first image requested

sytelus, 2017-11-29:
Does this happen for drone or car? Is the vehicle moving when taking first image? Can you please share any code that might help to reproduce this issue?

trbotha, 2017-11-29:
I'm using the data collection mode so this is with the drone. I'm basically using the StereoImageGenerator C++ example code provided. I have just moved the the pose generator and simSetPose to the beginning of the loop, which I also believe to be a bug in the original code as the pose is only set after the images are grabbed. I also tested with instead of a random pose a known translation pose to make it more evident.

In file _StereoImageGenerator.hpp_  under function ```int generate(int num_samples)```
In the code below ```ImageRequest(0, ImageType::Scene)``` will not have usable results. Note un-commenting this line seems to solve the problem ```//const std::vector<ImageResponse>& response1 = client.simGetImages(request);``` i.e. getting the images twice.

```pose_generator.next();
client.simSetPose(Pose(pose_generator.position, pose_generator.orientation), true);
std::vector<ImageRequest> request = {
	ImageRequest(0, ImageType::Scene),
	ImageRequest(1, ImageType::Scene),
	ImageRequest(2, ImageType::Scene),
	//ImageRequest(2, ImageType::DisparityNormalized, true)
	};
//const std::vector<ImageResponse>& response1 = client.simGetImages(request);
const std::vector<ImageResponse>& response = client.simGetImages(request);
if (response.size() != request.size()) {
	std::cout << "Images were not recieved!" << std::endl;
	start_nanos = clock->nowNanos();
	continue;
}
```

Jackyjae, 2017-12-13:
@trbotha hi, I've been working on this for several days but I can't figure out how to use simGetImages()
Can I use it directly in my actor?

trbotha, 2017-12-13:
@Jackyjae Yes, you can use the simGetImages() in your actor. I would advise you to look at the example code _StereoImageGenerator.hpp_. The code returns a vector response of images based on the requested images. The code below requests images from position 0,1 and 2. Believe that is center, left and right cameras.
```
std::vector<ImageRequest> request = {
					ImageRequest(0, ImageType::Scene), 
					ImageRequest(1, ImageType::Scene),
					ImageRequest(2, ImageType::Scene),
				};
const std::vector<ImageResponse>& response = client.simGetImages(request);
```

 I'm not sure what you wish to do with the images but you can access the raw image data as:
```
response.at(1).image_data_uint8.data()
```

where 1 denotes the second requested image location. Note that the data is compressed into jpeg format, not sure if this can be changed or not. You can use OpenCV to decode it into a cv::Mat and do more operations on it if you so desire

```
cv::Mat Left = cv::Mat(response.at(0).height, response.at(0).width, CV_8UC3);
Left = cv::imdecode(response.at(0).image_data_uint8, 1);
```
Hope this helps

sytelus, 2017-12-16:
I think this is now fixed in current master branch. The camera poses correctly comes out as 12.5cm on each side of center.

closed 2018-01-13 by sytelus