microsoft/AirSim #639: error at 98% while running ./build.sh, Ubuntu 16.04

dialogchat, 2017-11-28:
Scanning dependencies of target DroneServer
[ 98%] Building CXX object DroneServer/CMakeFiles/DroneServer.dir/home/root_alien/AirSim/DroneServer/main.cpp.o
In file included from /home/root_alien/AirSim/DroneServer/main.cpp:8:
/home/root_alien/AirSim/AirLib/include/vehicles/multirotor/controllers/RealMultirotorConnector.hpp:19:44: warning: 
      unused parameter 'dt' [-Wunused-parameter]
    virtual void updateRenderedState(float dt) override
                                           ^
/home/root_alien/AirSim/AirLib/include/vehicles/multirotor/controllers/RealMultirotorConnector.hpp:46:55: warning: 
      unused parameter 'index' [-Wunused-parameter]
    virtual VehicleCameraBase* getCamera(unsigned int index) override
                                                      ^
/home/root_alien/AirSim/AirLib/include/vehicles/multirotor/controllers/RealMultirotorConnector.hpp:52:38: warning: 
      unused parameter 'pose' [-Wunused-parameter]
    virtual void setPose(const Pose& pose, bool ignore_collision) override
                                     ^
/home/root_alien/AirSim/AirLib/include/vehicles/multirotor/controllers/RealMultirotorConnector.hpp:52:49: warning: 
      unused parameter 'ignore_collision' [-Wunused-parameter]
    virtual void setPose(const Pose& pose, bool ignore_collision) override
                                                ^
/home/root_alien/AirSim/AirLib/include/vehicles/multirotor/controllers/RealMultirotorConnector.hpp:62:61: warning: 
      unused parameter 'mesh_name' [-Wunused-parameter]
    virtual bool setSegmentationObjectID(const std::string& mesh_name, i...
                                                            ^
/home/root_alien/AirSim/AirLib/include/vehicles/multirotor/controllers/RealMultirotorConnector.hpp:62:76: warning: 
      unused parameter 'object_id' [-Wunused-parameter]
  ...bool setSegmentationObjectID(const std::string& mesh_name, int object_id,
                                                                    ^
/home/root_alien/AirSim/AirLib/include/vehicles/multirotor/controllers/RealMultirotorConnector.hpp:63:14: warning: 
      unused parameter 'is_name_regex' [-Wunused-parameter]
        bool is_name_regex = false) override
             ^
/home/root_alien/AirSim/AirLib/include/vehicles/multirotor/controllers/RealMultirotorConnector.hpp:67:60: warning: 
      unused parameter 'mesh_name' [-Wunused-parameter]
    virtual int getSegmentationObjectID(const std::string& mesh_name) override
                                                           ^
/home/root_alien/AirSim/DroneServer/main.cpp:80:29: error: variable type
      'msr::airlib::RealMultirotorConnector' is an abstract class
    RealMultirotorConnector connector(& mav_drone);
                            ^
/home/root_alien/AirSim/AirLib/include/controllers/VehicleConnectorBase.hpp:34:18: note: 
      unimplemented pure virtual method 'printLogMessage' in
      'RealMultirotorConnector'
    virtual void printLogMessage(const std::string& message, std::strin...
                 ^
8 warnings and 1 error generated.
DroneServer/CMakeFiles/DroneServer.dir/build.make:62: recipe for target 'DroneServer/CMakeFiles/DroneServer.dir/home/root_alien/AirSim/DroneServer/main.cpp.o' failed
make[2]: *** [DroneServer/CMakeFiles/DroneServer.dir/home/root_alien/AirSim/DroneServer/main.cpp.o] Error 1
CMakeFiles/Makefile2:556: recipe for target 'DroneServer/CMakeFiles/DroneServer.dir/all' failed
make[1]: *** [DroneServer/CMakeFiles/DroneServer.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

DavidLSmyth, 2017-11-28:
I had the same problem, have a look at pull request #638. The code was edited yesterday but it seems that one file was not updated that should have been.

ingenieroariel, 2017-11-29:
I  can confirm @DavidLSmyth's solution works.

dialogchat, 2017-11-29:
yeah, replace the old file with the new file from @DavidLSmyth' works. cheers.

closed 2017-11-29 by dialogchat

mindthink, 2017-12-01:
Thanks, it works for me