microsoft/AirSim #642: cannot open city environment binary

ldhblue3933, 2017-11-29:
[image]

 when i doubleclicked run.bat file in CityEnviron folder, i cannot open file with this error. how can i fix it?

n1ckfg, 2017-11-29:
Remove the line in settings.json where you set UAV or car, that fixed it for me. (Save a backup so you can restore after you're done trying the city demo.)

sytelus, 2017-12-01:
This is because we try to set stencil IDs for all the meshes at startup (this is required for segmentation view). City environment is pretty big and apparently render thread is timing out. In the new version I have put it out as 1.1.7 release, I've disabled setting stencil IDs. This means segmentation won't be available but you won't get this error for now.

ldhblue3933, 2017-12-11:
i have changed settings.json into initial one but failed to open city environment. but i can open it with version 1.1.7 release. i thank you for the reply and fixing it.

closed 2017-12-11 by ldhblue3933

lcyawym, 2019-05-08:
@sytelus  Hello, I download the cityEnViron. but I can't use 7.zip to extract in window.When I extract it , it occurs some error such as "header error". When I run the CityEnviron.sh. it occurs "error while loading shared libraries: libopenvr_api.so: cannot open shared object file: No such file or directory"