microsoft/AirSim #643: How to add and control more than one agent

rajnishti, 2017-11-30:
Hi I am doing path planning for multi vehicles using reinforcement learning. Can I add more than one agent in Unreal Engine environment?  If it is possible to add more agents in environment, then what are the changes that I should do.
Thanks

sytelus, 2017-12-01:
For drones its possible right now (see [this doc](https://github.com/Microsoft/AirSim/blob/master/docs/multi_vehicle.md)). For cars, this is under works.

saihv, 2017-12-24:
@sytelus Is that doc dated now? Because neither IsFpvVehicle nor VehicleConfigName are available for FlyingPawn

DanielLSM, 2018-01-30:
@sytelus Any updates on controlling multiple cars?

msb336, 2019-10-03:
Multiple car control is available as well. At this time, all vehicles in the scene must be the same simmode, but you can control multiple of whichever vehicle you choose.

closed 2019-10-03 by msb336