microsoft/AirSim #645: Build FAILED error C2259: 'msr::airlib::RealMultirotorConnector': cannot instantiate abstract class

tuzzer, 2017-11-30:
Build FAILED.

"C:\Projects\Roadrunner-Project\AirSim\AirSim.sln" (default target) (1) ->
"C:\Projects\Roadrunner-Project\AirSim\DroneServer\DroneServer.vcxproj" (default target) (6) ->
(ClCompile target) ->
  c:\projects\roadrunner-project\airsim\droneserver\main.cpp(80): error C2259: 'msr::airlib::RealMultirotorConnector':
cannot instantiate abstract class [C:\Projects\Roadrunner-Project\AirSim\DroneServer\DroneServer.vcxproj]

cangokalp, 2017-11-30:
I can't build either, getting the same error

XZ616313296, 2017-11-30:
I can't perform the behind because of this step either~~I am a novice

DavidLSmyth, 2017-11-30:
In AirLib/include/vehicles/multirotor/controllers/RealMultirotorConnector.hpp under public add the lines:
virtual void printLogMessage(const std::string& message, std::string message_param = "", unsigned char severity = 0) override
{
}

and then rebuild. See Pull request #638

commit ef987d5f (2017-12-01): Merge pull request #646 from tuzzer/master

Implement interface function printLogMessage to fix #645
closed 2017-12-01 by sytelus