microsoft/AirSim #649: Spoof Controller Input Programaticially

rjk2147, 2017-11-30:
Is there any way to send, from some python code, inputs to a flight controller that it would receive it and move exactly as it would as if it had received that input from a controller?

Has anyone worked on API controllers in AirSim?

sytelus, 2017-12-01:
Typically its not efficient to build controllers using Python on RPC APIs. If you had like to experiment with controllers, you should run AirSim from source code and directly modify code in AirLib. This is pretyy simple actually. AirSim uses flight controller called simple_flight which is pure C++11 code. It has [interface for controller](https://github.com/Microsoft/AirSim/blob/master/AirLib/include/vehicles/multirotor/firmwares/simple_flight/firmware/interfaces/IController.hpp). You just implement that interface and change [this line](https://github.com/Microsoft/AirSim/blob/master/AirLib/include/vehicles/multirotor/firmwares/simple_flight/firmware/Firmware.hpp#L75) to use your controller class instead of built-in CascadeController. The built-in [CascadeController](https://github.com/Microsoft/AirSim/blob/master/AirLib/include/vehicles/multirotor/firmwares/simple_flight/firmware/CascadeController.hpp) is simple cascaded PID controllers for rates, angle, velocity and position.

cangokalp, 2017-12-20:
@sytelus what did you mean by not efficient, in what way? 
I need a similar thing, to do this I was making use of the moveByManual function but  instead of RC input I send my inputs that my neural net produces. So I'd be basically using moveByManual and bypass RC input and replace it with my inputs. I just wrote a python method same as that moveByManual function in the c++ code but I just translated into python and instead of RC I make it take my inputs. Would this cause any issues?

cangokalp, 2017-12-20:
Follow up: What's the range of RC values that the moveByManual function expects?

closed 2018-02-13 by sytelus