microsoft/AirSim #650: Cannot compile on Linux

TobiasWeis, 2017-12-02:
Am not able to open the Blocks-World example after building UnrealEngine and AirSim. Have been following the README and the tutorial on how to open the Blocks-Environment on Linux. First it tells me that Blocks has been built with another version (as stated in the tutorial), I convert in place, then I need to rebuild, and here is where the following compilation errors occur:

`In file included from /home/weis/code/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Inc/AirSim/CameraDirector.gen.cpp:8:
In file included from ../../../Unreal/Environments/Blocks/Plugins/AirSim/Source/CameraDirector.h:4:
../../../Unreal/Environments/Blocks/Plugins/AirSim/Source/VehiclePawnWrapper.h:28:14: error: cannot use defaulted default constructor of
      'WrapperConfig' within 'VehiclePawnWrapper' outside of member functions because 'is_fpv_vehicle' has an initializer
        bool is_fpv_vehicle = false; 
             ^
../../../Unreal/Environments/Blocks/Plugins/AirSim/Source/VehiclePawnWrapper.h:29:21: error: cannot use defaulted default constructor of
      'WrapperConfig' within 'VehiclePawnWrapper' outside of member functions because 'vehicle_config_name' has an initializer
        std::string vehicle_config_name = ""; //use the default config name
                    ^
../../../Unreal/Environments/Blocks/Plugins/AirSim/Source/VehiclePawnWrapper.h:30:14: error: cannot use defaulted default constructor of
      'WrapperConfig' within 'VehiclePawnWrapper' outside of member functions because 'enable_collisions' has an initializer
        bool enable_collisions = true; 
             ^
../../../Unreal/Environments/Blocks/Plugins/AirSim/Source/VehiclePawnWrapper.h:31:14: error: cannot use defaulted default constructor of
      'WrapperConfig' within 'VehiclePawnWrapper' outside of member functions because 'enable_passthrough_on_collisions' has an initializer
        bool enable_passthrough_on_collisions = false; 
             ^
../../../Unreal/Environments/Blocks/Plugins/AirSim/Source/VehiclePawnWrapper.h:32:15: error: cannot use defaulted default constructor of
      'WrapperConfig' within 'VehiclePawnWrapper' outside of member functions because 'home_lattitude' has an initializer
        float home_lattitude = 47.641468;
              ^
../../../Unreal/Environments/Blocks/Plugins/AirSim/Source/VehiclePawnWrapper.h:33:15: error: cannot use defaulted default constructor of
      'WrapperConfig' within 'VehiclePawnWrapper' outside of member functions because 'home_longitude' has an initializer
        float home_longitude = -122.140165;
              ^
../../../Unreal/Environments/Blocks/Plugins/AirSim/Source/VehiclePawnWrapper.h:34:15: error: cannot use defaulted default constructor of
      'WrapperConfig' within 'VehiclePawnWrapper' outside of member functions because 'home_altitude' has an initializer
        float home_altitude = 122;
              ^
../../../Unreal/Environments/Blocks/Plugins/AirSim/Source/VehiclePawnWrapper.h:35:14: error: cannot use defaulted default constructor of
      'WrapperConfig' within 'VehiclePawnWrapper' outside of member functions because 'enable_trace' has an initializer
        bool enable_trace = false;
             ^
../../../Unreal/Environments/Blocks/Plugins/AirSim/Source/VehiclePawnWrapper.h:42:105: note: implicit default constructor for
      'VehiclePawnWrapper::WrapperConfig' first required here
    void initialize(APawn* pawn, const std::vector<APIPCamera*>& cameras, const WrapperConfig& config = WrapperConfig());`

reinaldomaslim, 2017-12-07:
I have the same problem

Kalay1995, 2017-12-07:
Same problem.

TobiasWeis, 2017-12-07:
I did not manage to fix it under linux. Switched to windows and had other compile errors. The commit before the refactoring ( 29f62755c57fb8f1de7c8aeb3e853698af22fe48 ) worked on windows for me.

Kalay1995, 2017-12-08:
I have fixed it, see #666.

sytelus, 2017-12-12:
This issue is now fixed in master. This is duplicate of https://github.com/Microsoft/AirSim/issues/666.

closed 2017-12-12 by sytelus