microsoft/AirSim #652: can no setting up Unreal environment

XinGao2017, 2017-12-03:
I use the freely downloadable environment(Landscape Mountain) and follow the step by step instructions to set up Unreal environment
Execute "Reopen LandscapeMountains.sln in Visual Studio, and make sure "DebugGame Editor" and "Win64" build configuration is the active build configuration." and then "Press F5 to run"

VS reports “This project is out of date：LandscapeMountains - DebugGame_Editor x64”
I choose not to build it

Then it pops up
“The following modules are missing or built with a different engien version:
UE4Editor-LandscapeMountains-Win64-DebugGame.dll
UE4Editor-AirSim-Win64-DebugGame.dll” 
I choose to rebuild them

That result the  error : "LandscapeMountains could not be compiled. Try rebuilding from source Manually".
Please help me , Thanks !

sytelus, 2017-12-06:
You need to choose to build it. If you get prompted to rebuild then chose yes.

XinGao2017, 2017-12-06:
When I choose to build it for the first prompt , then it reports as same as [https://github.com/Microsoft/AirSim/issues/587](url)
My system is Windows 7 x64 , so is it the cause ?
If so , I will install Windows 10 and try again .

XinGao2017, 2017-12-14:
The problem has been solved , it is caused by "Eigne/Dense" .

closed 2017-12-14 by XinGao2017

jjlee1991, 2018-02-01:
@sytelus Hi sir. I am facing the same errors. But mine did not face the issue of "Project out of date".

Followed the step by step, till the "F5" debug i faced the error
"The following modules are missing or built with a different engien version:
UE4Editor-LandscapeMountains-Win64-DebugGame.dll
Would you like to rebuild them now?"

Selected yes, and this error occurred:
"LandscapeMountains could not be compiled. Try rebuilding from source Manually".

Please advice
Thank you

samby707, 2018-12-20:
I am also facing the same issue. please help.