microsoft/AirSim #653: AirSim plugin compile error UE-4.18.1 on CentOS-7.4

edowson, 2017-12-03:
I am encountering the following error while attempting to build the AirSim plugin, copied over to my UE-4.18.1 project.

On my CentOS-7.4 installation, I'm using clang-5.0.1 and have build UE-4.18.1 (release branch) from sources. The default examples all work for cpp projects. I'm using Qt Creator 4.5 to build cpp projects.

UE-4.18 is built with clang-5.0 now, but AirSim current release supports only UE-4.17 and the setup script uses an older clang-3.9 version.

Could this issue be related to my use of clang-5.0.1? 

Any suggestions on where to start looking in order to modifying the cmake settings or build scripts to make the error go away?

```
Performing 3 actions (16 in parallel)
@progress 'Compiling C++ source code...' 0%
[1/3] Compile Module.AirSim.gen.cpp
[2/3] Compile Module.AirSim.cpp
In file included from /project/software/tutorial/airsim/airsim_quadrotor_lm/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.gen.cpp:5:
In file included from /project/software/tutorial/airsim/airsim_quadrotor_lm/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Inc/AirSim/CameraDirector.gen.cpp:8:
In file included from ../../../../tutorial/airsim/airsim_quadrotor_lm/Plugins/AirSim/Source/CameraDirector.h:4:
In file included from ../../../../tutorial/airsim/airsim_quadrotor_lm/Plugins/AirSim/Source/VehiclePawnWrapper.h:6:
In file included from ../../../../tutorial/airsim/airsim_quadrotor_lm/Plugins/AirSim/Source/VehicleCameraConnector.h:4:
In file included from ../../../../tutorial/airsim/airsim_quadrotor_lm/Plugins/AirSim/Source/PIPCamera.h:6:
In file included from /project/software/tutorial/airsim/airsim_quadrotor_lm/Plugins/AirSim/Source/AirLib/include/controllers/VehicleCameraBase.hpp:7:
In file included from /project/software/tutorial/airsim/airsim_quadrotor_lm/Plugins/AirSim/Source/AirLib/include/common/Common.hpp:16:
/project/software/tutorial/airsim/airsim_quadrotor_lm/Plugins/AirSim/Source/AirLib/include/common/VectorMath.hpp:9:1: error: 
unknown warning group '-Wmaybe-uninitialized', ignored [-Werror,-Wunknown-warning-option]
STRICT_MODE_OFF
^
/project/software/tutorial/airsim/airsim_quadrotor_lm/Plugins/AirSim/Source/AirLib/include/common/common_utils/StrictMode.hpp:67:5: note: expanded from macro 'STRICT_MODE_OFF'
    _Pragma("GCC diagnostic ignored \"-Wmaybe-uninitialized\"")         \
    ^
<scratch space>:351:25: note: expanded from here
 GCC diagnostic ignored "-Wmaybe-uninitialized"
                        ^
In file included from /project/software/tutorial/airsim/airsim_quadrotor_lm/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.cpp:3:
In file included from /project/software/tutorial/airsim/airsim_quadrotor_lm/Plugins/AirSim/Source/AirLib/src/api/RpcLibClientBase.cpp:10:
In file included from /project/software/tutorial/airsim/airsim_quadrotor_lm/Plugins/AirSim/Source/AirLib/include/api/RpcLibClientBase.hpp:7:
In file included from /project/software/tutorial/airsim/airsim_quadrotor_lm/Plugins/AirSim/Source/AirLib/include/common/Common.hpp:16:
/project/software/tutorial/airsim/airsim_quadrotor_lm/Plugins/AirSim/Source/AirLib/include/common/VectorMath.hpp:9:1: 
error: unknown warning group '-Wmaybe-uninitialized', ignored [-Werror,-Wunknown-warning-option]
STRICT_MODE_OFF
^
/project/software/tutorial/airsim/airsim_quadrotor_lm/Plugins/AirSim/Source/AirLib/include/common/common_utils/StrictMode.hpp:67:5: note: expanded from macro 'STRICT_MODE_OFF'
    _Pragma("GCC diagnostic ignored \"-Wmaybe-uninitialized\"")         \
    ^
<scratch space>:140:25: note: expanded from here
 GCC diagnostic ignored "-Wmaybe-uninitialized"
                        ^
In file included from /project/software/tutorial/airsim/airsim_quadrotor_lm/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.gen.cpp:5:
In file included from /project/software/tutorial/airsim/airsim_quadrotor_lm/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Inc/AirSim/CameraDirector.gen.cpp:8:
In file included from ../../../../tutorial/airsim/airsim_quadrotor_lm/Plugins/AirSim/Source/CameraDirector.h:4:
In file included from ../../../../tutorial/airsim/airsim_quadrotor_lm/Plugins/AirSim/Source/VehiclePawnWrapper.h:10:
/project/software/tutorial/airsim/airsim_quadrotor_lm/Plugins/AirSim/Source/AirLib/include/controllers/Settings.hpp:7:1: error: unknown warning group '-Wmaybe-uninitialized', ignored [-Werror,-Wunknown-warning-option]
STRICT_MODE_OFF
^
/project/software/tutorial/airsim/airsim_quadrotor_lm/Plugins/AirSim/Source/AirLib/include/common/common_utils/StrictMode.hpp:67:5: note: expanded from macro 'STRICT_MODE_OFF'
    _Pragma("GCC diagnostic ignored \"-Wmaybe-uninitialized\"")         \
    ^
<scratch space>:99:25: note: expanded from here
 GCC diagnostic ignored "-Wmaybe-uninitialized"
                        ^
In file included from /project/software/tutorial/airsim/airsim_quadrotor_lm/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.gen.cpp:5:
In file included from /project/software/tutorial/airsim/airsim_quadrotor_lm/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Inc/AirSim/CameraDirector.gen.cpp:8:
In file included from ../../../../tutorial/airsim/airsim_quadrotor_lm/Plugins/AirSim/Source/CameraDirector.h:4:
../../../../tutorial/airsim/airsim_quadrotor_lm/Plugins/AirSim/Source/VehiclePawnWrapper.h:42:105: error: default member initializer for 'is_fpv_vehicle' needed within definition of enclosing class 'VehiclePawnWrapper' outside of member functions
    void initialize(APawn* pawn, const std::vector<APIPCamera*>& cameras, const WrapperConfig& config = WrapperConfig());
                                                                                                        ^
../../../../tutorial/airsim/airsim_quadrotor_lm/Plugins/AirSim/Source/VehiclePawnWrapper.h:28:14: note: default member initializer declared here
        bool is_fpv_vehicle = false; 
             ^
../../../../tutorial/airsim/airsim_quadrotor_lm/Plugins/AirSim/Source/VehiclePawnWrapper.h:42:105: error: default member initializer for 'vehicle_config_name' needed within definition of enclosing class 'VehiclePawnWrapper' outside of member functions
[... 165 lines cut ...]
                                                                                                        ^
../../../../tutorial/airsim/airsim_quadrotor_lm/Plugins/AirSim/Source/VehiclePawnWrapper.h:35:14: note: default member initializer declared here
        bool enable_trace = false;
             ^
In file included from /project/software/tutorial/airsim/airsim_quadrotor_lm/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.cpp:24:
In file included from /project/software/tutorial/airsim/airsim_quadrotor_lm/Plugins/AirSim/Source/Multirotor/SimModeWorldMultiRotor.cpp:8:
In file included from /project/software/tutorial/airsim/airsim_quadrotor_lm/Plugins/AirSim/Source/AirLib/include/vehicles/multirotor/MultiRotorParamsFactory.hpp:9:
In file included from /project/software/tutorial/airsim/airsim_quadrotor_lm/Plugins/AirSim/Source/AirLib/include/vehicles/multirotor/configs/RosFlightQuadX.hpp:7:
/project/software/tutorial/airsim/airsim_quadrotor_lm/Plugins/AirSim/Source/AirLib/include/vehicles/multirotor/firmwares/ros_flight/RosFlightDroneController.hpp:17:1: error: unknown warning group '-Wmaybe-uninitialized', ignored [-Werror,-Wunknown-warning-option]
STRICT_MODE_OFF
^
/project/software/tutorial/airsim/airsim_quadrotor_lm/Plugins/AirSim/Source/AirLib/include/common/common_utils/StrictMode.hpp:67:5: note: expanded from macro 'STRICT_MODE_OFF'
    _Pragma("GCC diagnostic ignored \"-Wmaybe-uninitialized\"")         \
    ^
<scratch space>:101:25: note: expanded from here
 GCC diagnostic ignored "-Wmaybe-uninitialized"
                        ^
10 errors generated.
@progress 'Compiling C++ source code...' 25%
@progress 'Compiling C++ source code...' 50%
17 errors generated.
@progress 'Compiling C++ source code...' 75%
ERROR: UBT ERROR: Failed to produce item: /project/software/tutorial/airsim/airsim_quadrotor_lm/Plugins/AirSim/Binaries/Linux/libUE4Editor-AirSim.so
Total build time: 93.94 seconds (Local executor: 0.00 seconds)
```

edowson, 2017-12-03:
Apparently this issue is related to a bug llvm-config, and a workaround is to specify the -Wno-unknown-warning-option option:

https://stackoverflow.com/questions/41673546/clang-warning-warning-unknown-warning-option-wno-maybe-uninitialized

I tried adding the following entry to my qt-creator project for the UE project with the AirSim plugin files, but it appears to have no effect:

QMAKE_CXXFLAGS += -Wno-unknown-warning-option

I was able to build the plugin from the AirSim folder, after resolving some errors related to the system not being able to find libsupc++.a on CentOS-7.4 by disabling building of various executables. Libraries were built fine, but when you copy them over to a separate UE c++ project, it attempts to rebuild AirSim plugin again, and runs into the unknown -Wmaybe-uninitialized option error.

edowson, 2017-12-04:
I managed to resolve the compile issues with AirSim (master branch: commit  ef987d5f971d14ba9c73891089b8f635c0ca9ad4), UE-4.18.1 (4.18 branch: commit c009a7ad436601b404f3aef7f54914d6bca4a8bb), and Clang-5.0.1 built from source on CentOS 7.4 64-bit.

I haven't actually run the plugin, but UE-4.18.1 launches correctly from a Qt Creator 4.5 cpp project, with the AirSim plugin installed. (This is my first attempt at trying to work with AirSim).

Here are the patches. If it makes sense, I'll fork the repository and issue a pull request.

I'm surprised that the original code would even compile. For example, simple things like the following calls in CarPawn.cpp and VehiclePawn.cpp not having the correct number of arguments while calling VehiclePawnWrapper->initialize() (2, instead of 3), in the initializeForBeginPlay() methods (refer patches below, for the exact diffs):

```
void AFlyingPawn::initializeForBeginPlay()
{
    //get references of components so we can use later
    setupComponentReferences();

    std::vector<APIPCamera*> cameras = {fpv_camera_front_center_, fpv_camera_front_right_, fpv_camera_front_left_, 
        fpv_camera_bottom_center_, fpv_camera_back_center_};
    wrapper_->initialize(this, cameras, getVehiclePawnWrapper()->getConfig());
}
```

```
diff --git a/AirLib/include/common/common_utils/StrictMode.hpp b/AirLib/include/common/common_utils/StrictMode.hpp
index e38d6d5..d6a3029 100644
--- a/AirLib/include/common/common_utils/StrictMode.hpp
+++ b/AirLib/include/common/common_utils/StrictMode.hpp
@@ -64,7 +64,6 @@ _Pragma("clang diagnostic ignored \"-Wunused-variable\"")         \
     _Pragma("GCC diagnostic push")                                  \
     _Pragma("GCC diagnostic ignored \"-Wctor-dtor-privacy\"")        \
     _Pragma("GCC diagnostic ignored \"-Wdelete-non-virtual-dtor\"") \
-    _Pragma("GCC diagnostic ignored \"-Wmaybe-uninitialized\"")		\
     _Pragma("GCC diagnostic ignored \"-Wmissing-field-initializers\"") \
     _Pragma("GCC diagnostic ignored \"-Wold-style-cast\"")          \
     _Pragma("GCC diagnostic ignored \"-Wredundant-decls\"")         \
@@ -82,6 +81,7 @@ _Pragma("GCC diagnostic ignored \"-Wformat=\"")                 \
 _Pragma("GCC diagnostic ignored \"-Werror\"")                   \
 _Pragma("GCC diagnostic ignored \"-Werror=\"")                  \
 _Pragma("GCC diagnostic ignored \"-Wunused-variable\"")         \
+_Pragma("GCC diagnostic ignored \"-Wmaybe-uninitialized\"")		\
 */
 
 #define STRICT_MODE_ON                                            \
@@ -90,4 +90,4 @@ _Pragma("GCC diagnostic ignored \"-Wunused-variable\"")         \
 
 #endif
 
-#endif
\ No newline at end of file
+#endif
```

```
diff --git a/Unreal/Plugins/AirSim/Source/Car/CarPawn.cpp b/Unreal/Plugins/AirSim/Source/Car/CarPawn.cpp
index ed6aa83..824b2b1 100644
--- a/Unreal/Plugins/AirSim/Source/Car/CarPawn.cpp
+++ b/Unreal/Plugins/AirSim/Source/Car/CarPawn.cpp
@@ -206,7 +206,7 @@ void ACarPawn::initializeForBeginPlay(bool enable_rpc, const std::string& api_se
     setupInputBindings();
 
     std::vector<APIPCamera*> cameras = { InternalCamera1, InternalCamera2, InternalCamera3, InternalCamera4, InternalCamera5 };
-    wrapper_->initialize(this, cameras);
+    wrapper_->initialize(this, cameras, getVehiclePawnWrapper()->getConfig());
     wrapper_->setKinematics(&kinematics_);
 
     startApiServer(enable_rpc, api_server_address);
```

```
diff --git a/Unreal/Plugins/AirSim/Source/Multirotor/FlyingPawn.cpp b/Unreal/Plugins/AirSim/Source/Multirotor/FlyingPawn.cpp
index 3e6f946..a1101c6 100644
--- a/Unreal/Plugins/AirSim/Source/Multirotor/FlyingPawn.cpp
+++ b/Unreal/Plugins/AirSim/Source/Multirotor/FlyingPawn.cpp
@@ -18,7 +18,7 @@ void AFlyingPawn::initializeForBeginPlay()
 
     std::vector<APIPCamera*> cameras = {fpv_camera_front_center_, fpv_camera_front_right_, fpv_camera_front_left_, 
         fpv_camera_bottom_center_, fpv_camera_back_center_};
-    wrapper_->initialize(this, cameras);
+    wrapper_->initialize(this, cameras, getVehiclePawnWrapper()->getConfig());
 }
 
 void AFlyingPawn::NotifyHit(class UPrimitiveComponent* MyComp, class AActor* Other, class UPrimitiveComponent* OtherComp, bool bSelfMoved, FVector HitLocation, 
```

```
diff --git a/Unreal/Plugins/AirSim/Source/VehiclePawnWrapper.h b/Unreal/Plugins/AirSim/Source/VehiclePawnWrapper.h
index ea9a6ef..2b3345d 100644
--- a/Unreal/Plugins/AirSim/Source/VehiclePawnWrapper.h
+++ b/Unreal/Plugins/AirSim/Source/VehiclePawnWrapper.h
@@ -39,7 +39,7 @@ public:
 
 public: //interface
     VehiclePawnWrapper();
-    void initialize(APawn* pawn, const std::vector<APIPCamera*>& cameras, const WrapperConfig& config = WrapperConfig());
+    void initialize(APawn* pawn, const std::vector<APIPCamera*>& cameras, const WrapperConfig& config));
 
     void reset();
     void onCollision(class UPrimitiveComponent* MyComp, class AActor* Other, class UPrimitiveComponent* OtherComp, 
```

edowson, 2017-12-04:
The patches work and I'm able to run the AirSim quadrotor demo within a custom UE4 scene.

sytelus, 2017-12-06:
Thanks for looking into this. Yes, it would great if you can send pull request for the patch. I would be happy to merge with the master.

edowson, 2017-12-11:
I have applied the patches and created a pull request.
#676

sytelus, 2017-12-12:
I've [merged](https://github.com/Microsoft/AirSim/commit/23e11d9e6568ac4192f2a849dba5cb3dc1efa50d) pull request (with different approach to fix for VehiclwPawnWrapper).

closed 2017-12-12 by sytelus