microsoft/AirSim #654: Frequent Disconnection on MultirotorClient

maximilianchang, 2017-12-03:
I am running the neighborhood environment in AirSim and having a Multirotor Client do a random walk in the environment. I find that the drone will often stop responding to commands with the requests timing out. I am wondering if anyone else has encountered a similar issue. 

I am running on Windows environment.

sytelus, 2017-12-06:
Are you using default simple_flight or PX4? Could you please give any script that helps reproduce your issue? I would be also interested in knowing your machine config (GPU, CPU, RAM).

maximilianchang, 2017-12-06:
I'm using simple_flight and doing a simple random walk (i.e. making calls to _moveToPosition(current_position.x_val + randint(0, 5), current_position.y_val + randint(0, 5), current_position.z_val, 5)_ ). 

I'm using a GPU (NVIDIA Tesla M60) with 8GiB of GPU memory. More specifically, this is the G3 Instance Type offered on AWS.

sytelus, 2018-01-13:
Please provide full code to reproduce this issue. When running locally, I'm not seeing this issue so I think this might be more related to your AWS setup.

closed 2018-02-13 by sytelus