microsoft/AirSim #655: RPCLib not found at build time

Jazzinghen, 2017-12-04:
I have had issues with RPCLib not being found by the build script under Windows 10. The script is set to look for the library in `deps\rpclib\lib\x64\Debug`, however that folder does not exist.
I found an issue in the build script where it copies RPCLib:
`robocopy /MIR external\rpclib\build\output\lib %RPCLIB_TARGET_LIB%`
I have checked and the RPCLib binaries are located either in `external\rpclib\build\Release` or `external\rpclib\build\Debug`.
I have managed to build everything by manually copying the two folders in `deps\rpclib\lib\x64`.

sytelus, 2017-12-06:
I just deleted my build folder and ran build and again and the binaries show up in `external\rpclib\build\output\lib` which the gets correctly copied to `deps\rpclib\lib\x64\`. 

Are there any differences in the way you are doing the build? Are you using VS 2015?

Jazzinghen, 2017-12-06:
No, I am using Visual Studio 2017, I had to swap the two cmake calls for RPCLib:
```
cmake -G"Visual Studio 15 2017 Win64" ..
REM cmake -G"Visual Studio 14 2015 Win64" ..
```

sytelus, 2017-12-06:
That might be the reason. I would suggest to do build from "Developer command Prompt for VS 2017". If you do that then you won't need any changes in build.cmd and things should just work out of box.

Jazzinghen, 2017-12-06:
I was already executing `build.cmd` from the Developer Command Prompt for VS 2017. I already tried building without changing that line as suggested in another bug report, but I get errors during RPCLib build:
```
    C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.Cpp.Platform.targets(57,5): error MSB8020: The build tools for v140 (Platform Toolset = 'v140') cannot be found. To build using the v140 build tools, please install v140 build tools.  Alternatively, you may upgrade to the current Visual Studio tools by selecting the Project menu or right-click the solution, and then selecting "Retarget solution". [C:\Users\Jazzinghen\Documents\GitHub\AirSim\external\rpclib\build\CMakeFiles\3.9.6\VCTargetsPath.vcxproj]
    Done Building Project "C:\Users\Jazzinghen\Documents\GitHub\AirSim\external\rpclib\build\CMakeFiles\3.9.6\VCTargetsPath.vcxproj" (default targets) -- FAILED.
```

I have only Visual Studio 2017 installed on my PC.

sytelus, 2018-02-13:
We'll be supporting VS2017 in near future.

closed 2018-02-13 by sytelus