microsoft/AirSim #659: Collision detection

celloaerometer, 2017-12-04:
Hi folks,

I have some troubles with the collision detection feature.
Whenever I play with my custom car or with the default rc car I don't get any collision events (such as 'Onhit', 'ReceiveActorBeginOverlap', 'NotifyHit' and etc).

I tried to use the 'getCollisionInfo' method from the airsimclient.py file. But I always get only has_collided set to false.

Is anyone here have any suggestions??

sytelus, 2017-12-05:
You likely don't have either collision mesh on your car and/or haven't enabled the collision on it. How you do this depends on how you made your custom car and you should refer to Unreal documentation or forums. In our case, we had to go to car's physics asset, chose ChasisBone, go to Collision section and check "Simulation Generates Hit Events" box.

celloaerometer, 2017-12-06:
Hi Shital,

Thanks! you were right.
"Simulation Generates Hit Events" marked only in the car's bp and not in the physics asset. now it's working well! :)

closed 2017-12-06 by celloaerometer