microsoft/AirSim #660: ClockSpeed setting has no effect

valtsblukis, 2017-12-05:
I've attempted to set the ClockSpeed setting in settings.json as explained in the docs, but this seems to not have any effect on the actual speed of the simulation. It is correctly read by AirSim, as evidenced by the on-screen output after startup. Regardless of the value I set, the drone is still flying at the same velocity. I would have assumed that with ClockSpeed: 3.0, a velocity command should result in the drone appearing to fly 3 times faster compared to real-time. Is this wrong?

Thanks!

sytelus, 2017-12-05:
Currently clock speed applies to only car physics. There is all internal infrastructure to apply to drones as well but that part hasn't been implemented and tested yet.

valtsblukis, 2017-12-06:
I see. I wouldn't mind giving the implementation a go, if it's not terribly complicated, because I really need this for my project. Could you point me in the right direction? Any prior commits to refer to, or which files to modify? Thanks!

sytelus, 2017-12-06:
I have added the change. It turned out to be pretty simple (please see above commit).

However you would probably want to know some pro and cons. When you increase the clock speed, you are in essence telling physics engine to use longer periods to compute new physics state. This means as you speed up the clock, the quality of physics simulation reduces and things starts diverging fast. This is especially impactful for drones (than cars) because drones require high frequency stabilization loop. So as you speed up the clock, drone would become more unstable.

valtsblukis, 2017-12-06:
Thanks a lot, Shital! I appreciate your help and I understand the implications. My hardware allows me to increase the physics rate a bit, so I have some room to play with the clock rate.

valtsblukis, 2017-12-06:
Just reporting my results. I managed to successfully set ClockSpeed: 10 and fly the drone just fine at a control rate of 5Hz (effectively 50Hz). The traced out trajectories are slightly different than at ClockSpeed: 1, but still very good.

To do this, I first increased the physics rate 30 times to 0.1ms / 10'000Hz. At this point the drone was still unstable if ClockSpeed > 3.0, which seemed strange. I found that PidController uses the wall clock to get the time delta, but if ClockSpeed != 1, then wall clock is inconsistent with the physics. This was causing the PidController to overcorrect and become unstable. So I multiplied the dt by ClockSpeed inside the PidController (patch attached) and now the drone is stable at any ClockSpeed < 30.

Finally, I had to make sure that the unreal FPS is higher than the rate at which I'm requesting camera images, and I also set the AutoExposureSpeed to 1, otherwise the images were too bright.

[0001-Modified-PidController-to-take-into-account-ClockSpe.patch.txt](https://github.com/Microsoft/AirSim/files/1536734/0001-Modified-PidController-to-take-into-account-ClockSpe.patch.txt)

cangokalp, 2017-12-21:
@valtsblukis This is awesome!

Xking8, 2018-01-10:
@valtsblukis, I got some issues to speedup ClockSpeed. I followed your steps but still not working.
When the ClockSpeed <2, it is okay. When the ClockSpeed >=2, the drone is unstable. Even more, when ClockSpeed >=3, the drone hit into ground right after it take off.
However, i wish to speedup for a least 5 times.
Are these issues because of my computation power? I'm running on a PC (CPU: intel i7-7700@3.6GHz, RAM:16G, GPU: GTX1070)
Have any advises for me to speedup my simulation? 

Thanks!

guillem74, 2018-04-02:
@sytelus Is there some advance regarding drone speed up stability?

sytelus, 2018-04-13:
Just fixed this issue. On a powerful desktop, I can run quadcopter simulation now at 100X clock speed!

Related: https://github.com/Microsoft/AirSim/issues/901

closed 2018-04-13 by sytelus

tvturnhout, 2018-10-09:
> Just fixed this issue. On a powerful desktop, I can run quadcopter simulation now at 100X clock speed!
> 
> Related: #901

Could you share the specs of the desktop?