microsoft/AirSim #661: Unreal Engine Build Error

satrialoka, 2017-12-06:
when i run ./Setup.sh to build the UnrealEngine i got this 
any suggestion how to fix this ?

`dpkg-query: no packages found matching clang-3.8
Attempting installation of missing package: clang-3.8
+ sudo apt-get install -y clang-3.8
[sudo] password for lokasatria: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 clang-3.8 : Depends: libclang-common-3.8-dev (= 1:3.8-2ubuntu1) but it is not going to be installed
             Recommends: llvm-3.8-dev but it is not going to be installed
E: Unable to correct problems, you have held broken packages.`

sytelus, 2017-12-12:
I suspect you have broken packages and apt-get is refused to go ahead. You should search on how to fix packages on your Linux system and then try again.

closed 2017-12-12 by sytelus

carsontang, 2018-01-26:
@satrialoka, did you ever find a solution to this? I'm encountering the same issue.