microsoft/AirSim #662: AirSim/HelloDrone/main.cpp:24:13: error: use of undeclared identifier 'VehicleCameraBase'?

kecaiwu, 2017-12-06:
/home/cc/AirSim/HelloDrone/main.cpp:24:13: error: use of undeclared identifier 'VehicleCameraBase'
    typedef VehicleCameraBase::ImageRequest ImageRequest;
            ^
/home/cc/AirSim/HelloDrone/main.cpp:25:13: error: use of undeclared identifier 'VehicleCameraBase'
    typedef VehicleCameraBase::ImageResponse ImageResponse;
            ^
/home/cc/AirSim/HelloDrone/main.cpp:26:13: error: use of undeclared identifier 'VehicleCameraBase'
    typedef VehicleCameraBase::ImageType ImageType;
            ^
/home/cc/AirSim/HelloDrone/main.cpp:33:58: error: 'ImageType' (aka 'int') is not a class, namespace, or enumeration
        vector<ImageRequest> request = { ImageRequest(0, ImageType::Scene), ImageRequest(1, ImageType::DepthPlanner, true) };
                                                         ^
/home/cc/AirSim/HelloDrone/main.cpp:33:93: error: 'ImageType' (aka 'int') is not a class, namespace, or enumeration
        vector<ImageRequest> request = { ImageRequest(0, ImageType::Scene), ImageRequest(1, ImageType::DepthPlanner, true) };
                                                                                            ^
/home/cc/AirSim/HelloDrone/main.cpp:34:69: error: no viable conversion from 'vector<ImageRequest>' to 'vector<ImageCaptureBase::ImageRequest>'
        const vector<ImageResponse>& response = client.simGetImages(request);
                                                                    ^~~~~~~
/home/cc/AirSim/llvm-build/output/include/c++/v1/vector:538:5: note: candidate constructor not viable: no known conversion from 'vector<ImageRequest>'
      (aka 'std::__1::vector<int, std::__1::allocator<int> >') to 'initializer_list<value_type>' (aka 'initializer_list<msr::airlib::ImageCaptureBase::ImageRequest>') for 1st argument
    vector(initializer_list<value_type> __il);
    ^
/home/cc/AirSim/llvm-build/output/include/c++/v1/vector:550:5: note: candidate constructor not viable: no known conversion from 'vector<ImageRequest>'
      (aka 'std::__1::vector<int, std::__1::allocator<int> >') to 'const std::__1::vector<msr::airlib::ImageCaptureBase::ImageRequest,
      std::__1::allocator<msr::airlib::ImageCaptureBase::ImageRequest> > &' for 1st argument
    vector(const vector& __x);
    ^
/home/cc/AirSim/llvm-build/output/include/c++/v1/vector:556:5: note: candidate constructor not viable: no known conversion from 'vector<ImageRequest>'
      (aka 'std::__1::vector<int, std::__1::allocator<int> >') to 'std::__1::vector<msr::airlib::ImageCaptureBase::ImageRequest,
      std::__1::allocator<msr::airlib::ImageCaptureBase::ImageRequest> > &&' for 1st argument
    vector(vector&& __x)
    ^
/home/cc/AirSim/AirLib/include/api/RpcLibClientBase.hpp:24:97: note: passing argument to parameter 'request' here
    vector<ImageCaptureBase::ImageResponse> simGetImages(vector<ImageCaptureBase::ImageRequest> request);
                                                                                                ^
/home/cc/AirSim/HelloDrone/main.cpp:43:64: error: member reference base type 'const ImageResponse' (aka 'const int') is not a structure or union
                std::cout << "Image uint8 size: " << image_info.image_data_uint8.size() << std::endl;
                                                     ~~~~~~~~~~^~~~~~~~~~~~~~~~~
/home/cc/AirSim/HelloDrone/main.cpp:44:64: error: member reference base type 'const ImageResponse' (aka 'const int') is not a structure or union
                std::cout << "Image float size: " << image_info.image_data_float.size() << std::endl;
                                                     ~~~~~~~~~~^~~~~~~~~~~~~~~~~
/home/cc/AirSim/HelloDrone/main.cpp:47:96: error: member reference base type 'const ImageResponse' (aka 'const int') is not a structure or union
                    std::string file_path = FileSystem::combine(path, std::to_string(image_info.time_stamp));
                                                                                     ~~~~~~~~~~^~~~~~~~~~~
/home/cc/AirSim/HelloDrone/main.cpp:48:35: error: member reference base type 'const ImageResponse' (aka 'const int') is not a structure or union
                    if (image_info.pixels_as_float) {
                        ~~~~~~~~~~^~~~~~~~~~~~~~~~
fatal error: too many errors emitted, stopping now [-ferror-limit=]
11 errors generated.
HelloDrone/CMakeFiles/HelloDrone.dir/build.make:62: recipe for target 'HelloDrone/CMakeFiles/HelloDrone.dir/home/cc/AirSim/HelloDrone/main.cpp.o' failed
make[2]: *** [HelloDrone/CMakeFiles/HelloDrone.dir/home/cc/AirSim/HelloDrone/main.cpp.o] Error 1
CMakeFiles/Makefile2:385: recipe for target 'HelloDrone/CMakeFiles/HelloDrone.dir/all' failed
make[1]: *** [HelloDrone/CMakeFiles/HelloDrone.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

spsancti, 2017-12-06:
Faced the same issue. Is there any workaround now?

sytelus, 2017-12-06:
Fixed.

closed 2017-12-06 by sytelus