microsoft/AirSim #663: Build error

Jackyjae, 2017-12-06:
Updating submodules...
fatal: Not a git repository (or any of the parent directories): .git
Starting cmake...
-- Selecting Windows SDK version  to target Windows 10.0.15063.
CMake Error at CMakeLists.txt:2 (project):
  Failed to run MSBuild command:

    MSBuild.exe

  to get the value of VCTargetsPath:

    Microsoft (R) Build Engine version 15.4.8.50001 for .NET Framework
    Copyright (C) Microsoft Corporation. All rights reserved.

    Build started 2017/12/6 17:23:44.
    Project "D:\Jack\AirSim-1.1.7\external\rpclib\build\CMakeFiles\3.10.0\VCTargetsPath.vcxproj" on node 1 (default targets).
    D:\IDE\VS\Common7\IDE\VC\VCTargets\Microsoft.Cpp.Platform.targets(57,5): error MSB8020: The build tools for v140 (Platform Toolset = 'v140') cannot be found. To build using the v140 build tools, please install v140 build tools.  Alternatively, you may upgrade to the current Visual Studio tools by selecting the Project menu or right-click the solution, and then selecting "Retarget solution". [D:\Jack\AirSim-1.1.7\external\rpclib\build\CMakeFiles\3.10.0\VCTargetsPath.vcxproj]
    Done Building Project "D:\Jack\AirSim-1.1.7\external\rpclib\build\CMakeFiles\3.10.0\VCTargetsPath.vcxproj" (default targets) -- FAILED.

    Build FAILED.

    "D:\Jack\AirSim-1.1.7\external\rpclib\build\CMakeFiles\3.10.0\VCTargetsPath.vcxproj" (default target) (1) ->
    (PlatformPrepareForBuild target) ->
      D:\IDE\VS\Common7\IDE\VC\VCTargets\Microsoft.Cpp.Platform.targets(57,5): error MSB8020: The build tools for v140 (Platform Toolset = 'v140') cannot be found. To build using the v140 build tools, please install v140 build tools.  Alternatively, you may upgrade to the current Visual Studio tools by selecting the Project menu or right-click the solution, and then selecting "Retarget solution". [D:\Jack\AirSim-1.1.7\external\rpclib\build\CMakeFiles\3.10.0\VCTargetsPath.vcxproj]

        0 Warning(s)
        1 Error(s)

    Time Elapsed 00:00:00.10

  Exit code: 1

-- Configuring incomplete, errors occurred!
See also "D:/Jack/AirSim-1.1.7/external/rpclib/build/CMakeFiles/CMakeOutput.log".
Microsoft (R) Build Engine version 15.4.8.50001 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

MSBUILD : error MSB1009: Project file does not exist.
Switch: ALL_BUILD.vcxproj
Microsoft (R) Build Engine version 15.4.8.50001 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

MSBUILD : error MSB1009: Project file does not exist.
Switch: ALL_BUILD.vcxproj
#### Build failed

D:\Jack\AirSim-1.1.7>build.cmd
Updating submodules...
fatal: Not a git repository (or any of the parent directories): .git
Starting cmake...
-- Selecting Windows SDK version  to target Windows 10.0.15063.
CMake Error at CMakeLists.txt:2 (project):
  Failed to run MSBuild command:

    MSBuild.exe

  to get the value of VCTargetsPath:

    Microsoft (R) Build Engine version 15.4.8.50001 for .NET Framework
    Copyright (C) Microsoft Corporation. All rights reserved.

    Build started 2017/12/6 17:43:43.
    Project "D:\Jack\AirSim-1.1.7\external\rpclib\build\CMakeFiles\3.10.0\VCTargetsPath.vcxproj" on node 1 (default targets).
    D:\IDE\VS\Common7\IDE\VC\VCTargets\Microsoft.Cpp.Platform.targets(57,5): error MSB8020: The build tools for v140 (Platform Toolset = 'v140') cannot be found. To build using the v140 build tools, please install v140 build tools.  Alternatively, you may upgrade to the current Visual Studio tools by selecting the Project menu or right-click the solution, and then selecting "Retarget solution". [D:\Jack\AirSim-1.1.7\external\rpclib\build\CMakeFiles\3.10.0\VCTargetsPath.vcxproj]
    Done Building Project "D:\Jack\AirSim-1.1.7\external\rpclib\build\CMakeFiles\3.10.0\VCTargetsPath.vcxproj" (default targets) -- FAILED.

    Build FAILED.

    "D:\Jack\AirSim-1.1.7\external\rpclib\build\CMakeFiles\3.10.0\VCTargetsPath.vcxproj" (default target) (1) ->
    (PlatformPrepareForBuild target) ->
      D:\IDE\VS\Common7\IDE\VC\VCTargets\Microsoft.Cpp.Platform.targets(57,5): error MSB8020: The build tools for v140 (Platform Toolset = 'v140') cannot be found. To build using the v140 build tools, please install v140 build tools.  Alternatively, you may upgrade to the current Visual Studio tools by selecting the Project menu or right-click the solution, and then selecting "Retarget solution". [D:\Jack\AirSim-1.1.7\external\rpclib\build\CMakeFiles\3.10.0\VCTargetsPath.vcxproj]

        0 Warning(s)
        1 Error(s)

    Time Elapsed 00:00:00.12

  Exit code: 1

sytelus, 2017-12-06:
Are you using VS2015? In that case, are you running build from VS2015 Native Tools x64 Command Prompt?

Jackyjae, 2017-12-07:
I installed both vs15 and 17. I did use VS2015 Native Tools x64 Command Prompt to run build.cmd. Is it because of the CMake version? Mine is 3.10.

Jackyjae, 2017-12-07:
I clone the AirSim in the VS2017 Native Tools x64 Command Prompt agian and it works now.

closed 2017-12-07 by Jackyjae

MeditationG, 2017-12-18:
I have the same issue to you! I follow the guidance to build but it still does not work

ferhatmelih, 2019-01-07:
For anyone still having the issue, I just solved the problem like this,
Turns out when you run Native Tools Command Prompt as Administrator, the default location you end up in is
$ C:\Windows\System32

However, you want to clone and build AirSim in 
$ C:\Program Files\Visual Studio 2017\Community

When you follow the rest of the steps, 
$cd AirSim 
$build.cmd
It works out perfectly.