microsoft/AirSim #664: Update multi_vehicle.md with python API example

DavidLSmyth, 2017-12-06:
Would it be possible to update multi_vehicle.md to reflect recent changes? For example, there is no longer a VehicleConfig section under BP_Flying_Pawn details. An full example of creating multiple vehicles and controlling them via the python API on separate ports with simple flight as well as PX4 firmware would be highly appreciated. 

I have tried to run AirSim with multiple vehicles in the environment but have only ever managed to get the first listed in settings.json to connect via the python API. Opening a fresh issue because I couldn't figure out the correct configuration from #230

saihv, 2017-12-06:
Are you already instantiating multiple vehicles with different API ports? If yes, I have a really barebones script that can send commands to multiple drones at my fork, which you can probably extend based on the functionality you need:

https://github.com/saihv/AirSim/blob/8e6fad6abfeb8e01da482638407a8a6badc09ef0/PythonClient/swarmControl.py

+1 for an official Python API as well!

DavidLSmyth, 2017-12-06:
No, I cannot instantiate multiple vehicles with different API ports yet. I outlined some problems I was having in #230. I have a similar script to yours which should be able to control the drones once they are instantiated, I'm just having trouble getting drones that are not marked as Is Fpv Vehicle to create a socket to connect to

DavidLSmyth, 2017-12-06:
@saipv I cloned your fork and managed to instantiate multiple vehicles running on separate ports, if someone could advise how to deal with the missing "Vehicle Config Name" in the latest version of AirSim that would be very helpful.

saipv, 2017-12-06:
Notifying @saihv as @DavidLSmyth might have mistakenly looped me in :)

saihv, 2017-12-09:
@DavidLSmyth 

Perhaps an ugly way of doing it, but you can probably 'append' the index of the vehicle to the new wrapper structure that is in AirSim, in order to differentiate between vehicles and use multiple API ports.: I believe the reason for the wrapper is to use the same method for instantiating both cars and drones. The following is a modification of the code from SimModeMultiRotor.cpp

`
unsigned int vehicleCount = 0;
        //set up vehicle pawns
        for (AActor* pawn : pawns)
        {
            //initialize each vehicle pawn we found
            TMultiRotorPawn* vehicle_pawn = static_cast<TMultiRotorPawn*>(pawn);
            vehicle_pawn->initializeForBeginPlay();

            //chose first pawn as FPV if none is designated as FPV
            VehiclePawnWrapper* wrapper = vehicle_pawn->getVehiclePawnWrapper();

	 <mark>   wrapper->getConfig().vehicle_config_name = "Vehicle" + std::to_string(vehicleCount); </mark>
			
            if (enable_collision_passthrough)
                wrapper->getConfig().enable_passthrough_on_collisions = true;
            if (wrapper->getConfig().is_fpv_vehicle || fpv_vehicle_pawn_wrapper_ == nullptr)
                fpv_vehicle_pawn_wrapper_ = wrapper;

            //now create the connector for each pawn
            VehiclePtr vehicle = createVehicle(wrapper);
            if (vehicle != nullptr) {
                vehicles.push_back(vehicle);

                if (fpv_vehicle_pawn_wrapper_ == wrapper)
                    fpv_vehicle_connector_ = vehicle;
            }
            //else we don't have vehicle for this pawn
        }

`

saihv, 2017-12-09:
@DavidLSmyth 

Perhaps an ugly way of doing it, but in the newest version, you can probably 'append' the index of the vehicle to the new wrapper structure that is in AirSim, in order to differentiate between vehicles and use multiple API ports.: I believe the reason for the wrapper is to use the same method for instantiating both cars and drones. The following is a quick modification of the code from SimModeMultiRotor.cpp, and I think this can be extended to create custom numbering..
```

unsigned int vehicleCount = 0;
        //set up vehicle pawns
        for (AActor* pawn : pawns)
        {
            //initialize each vehicle pawn we found
            TMultiRotorPawn* vehicle_pawn = static_cast<TMultiRotorPawn*>(pawn);
            vehicle_pawn->initializeForBeginPlay();

            //chose first pawn as FPV if none is designated as FPV
            VehiclePawnWrapper* wrapper = vehicle_pawn->getVehiclePawnWrapper();

	   wrapper->getConfig().vehicle_config_name = "Vehicle" + std::to_string(vehicleCount);
           vehicleCount++;
			
            if (enable_collision_passthrough)
                wrapper->getConfig().enable_passthrough_on_collisions = true;
            if (wrapper->getConfig().is_fpv_vehicle || fpv_vehicle_pawn_wrapper_ == nullptr)
                fpv_vehicle_pawn_wrapper_ = wrapper;

            //now create the connector for each pawn
            VehiclePtr vehicle = createVehicle(wrapper);
            if (vehicle != nullptr) {
                vehicles.push_back(vehicle);

                if (fpv_vehicle_pawn_wrapper_ == wrapper)
                    fpv_vehicle_connector_ = vehicle;
            }
            //else we don't have vehicle for this pawn
        }
```

And then of course, you'd have to match these Vehicle0, Vehicle1 tags with stuff in settings.json

Looks like the difference between the old and new versions is that the new version has a wrapper structure, which stores vehicle configuration data internally, but the older versions had a UPROPERTY that allowed you to change the fields from Unreal.

saihv, 2017-12-24:
Fixed it: after the API ports are changed, startApiServer() needs to be called for each vehicle in the list. Which means fpv_vehicle_connector needs to be an array of pointers instead of just one:

https://gist.github.com/saihv/1611ae5d34fbdb52c814efb1023eb154

DavidLSmyth, 2017-12-24:
That makes sense, I had been working on this and noticed that the connections weren't being created but didn't have time to figure it out, thanks for sharing the fix. I think this should be submitted as a pull request, it's listed as p0 on the trello: https://trello.com/b/1t2qCeaA/todo

saihv, 2017-12-25:
Ah, I didn't even notice there was a Trello! I will wrap my changes up into a commit and submit it.

SaiVineethJ, 2018-02-15:
@saihv Can u tell a sample command for swarm.py using keyboard after initializing the drone??

jixian79, 2018-06-19:
@saihv How can I set the subwidows for Multiple Vehicles?

zhuboshun2015, 2019-01-15:
Hello, I'm a new AirSim user. I have a question about multiple vehicles. Now I can add multiple cars in the Moutains Landscape Evenvironment , but I can just control only on car while the other one accepts the same control . I want to achieve that the "host" car is controled and the other cars run around the "host " car . How can I achieve this ? @saihv

zimmy87, 2021-03-12:
closed to age and lack of activity. this also looks like it's fixed at this point

closed 2021-03-12 by zimmy87