microsoft/AirSim #665: Request time out

cangokalp, 2017-12-06:
This is related to the #631 issue that has been closed recently;

For me the issue changed and now I get this after some time randomly . Similar to the previous issue but now instead of crashing this happens;
self.client.call('reset')
File "...\Anaconda3\lib\site-packages\msgpackrpc\session.py", line 41, in call
return self.send_request(method, args).get()
File "..\Anaconda3\lib\site-packages\msgpackrpc\future.py", line 43, in get
raise self._error
msgpackrpc.error.TimeoutError: Request timed out

Kjell-K, 2017-12-06:
@cangokalp @sytelus +1 unfortunately. Unreal Window is still open with the quadcopter in the terminal state (collision) for reset. This happened after 125k steps and 4k successful resets. So therefore very randomly as well.

For now, I will build the reset into a while loop to make sure home position is reached. Maybe a possible workaround.

sytelus, 2017-12-06:
Request timeout may happen because of state of previous command. I would suggest to handle this exception and figure out which request/command (such as move*) caused it. More information, better it would be for debugging....

Kjell-K, 2017-12-07:
After I crash I return that info and before I call the client.reset (where the timeout error is raised), I perform:
`now = client.getPosition()`
and 
`responses = client.simGetImages([ImageRequest(0, AirSimImageType.DepthPerspective, True, False)])`

But the question is why this occurs since the #631 fix.
Is there a suggested exception handling for time out especially for AirSim?

cangokalp, 2017-12-08:
Yeah for me it occurs after client.getPosition (before that moveByVelocityZ is called). And it randomly happens again. So random crash issue turned in to a random timeout error after calling reset() - after #631

Kjell-K, 2017-12-08:
But since you are calling reset, do you had a collision when the error occurs? In my case it is always connected to a collision. Then we might want to try not to call anything in-between collision and reset.

cangokalp, 2017-12-08:
Not for me it's not after a collision.

Kjell-K, 2017-12-09:
Now I had an instance without a previous collision. 

@sytelus It ALWAYS occurs at client.reset. I do not think the previous command is causing it since I changed it up and call reset from different previous commands. Could the reset itself cause a break of connection to Unreal?

cangokalp, 2018-01-12:
Unfortunately this issue still persist, 

@Kjell-K do you have any temporary solution for this?

I can't get my training go for a long time as either this timeout error occurs or it throws an exception randomly ( #701 ). I think those are the two biggest problems at the moment - It's not possible to train an RL agent more than a day, as simulation is crashing due to one of these issues.

Kjell-K, 2018-01-12:
@cangokalp Yes my temporary solution is to return to an old AirSim version where I can use setPose() to suddenly move the copter. I customized my own reset with this. The limitation is that the copter cannot go through obstacles (therefore not suited for every environment).  

It really is a work around but runs stable. I trained for a 5 days without issues.

Still waiting for a fix of this issue post in order to move to the newest AirSim version.

cangokalp, 2018-01-23:
The problem might be in msgpackrpc

```
..., in _reset
    self.airsimClient.enableApiControl(True)
.., line 194, in enableApiControl
    return self.client.call('enableApiControl', is_enabled)
  File "C:\Users\modeler\Anaconda3\lib\site-packages\msgpackrpc\session.py", line 41, in call
    return self.send_request(method, args).get()
  File "C:\Users\modeler\Anaconda3\lib\site-packages\msgpackrpc\future.py", line 43, in get
    raise self._error
msgpackrpc.error.TimeoutError: Request timed out
```

or for some reason it does it, I'll post the call stack next time it happens.

sytelus, 2018-02-13:
I'd done major refactoring of image capture subsystem. Have you tried new version?

cangokalp, 2018-02-15:
With the new version - I think the issue is gone (at least for me). I don't experience it anymore. @Kjell-K you might want to check it out too. Thanks @sytelus

closed 2018-02-15 by cangokalp

tachitachi, 2018-02-21:
I'm still getting this issue. And after the timeout error, it's not possible to reconnect again without restarting the Unreal client.

paulfauthmayer, 2018-06-27:
I'm having the same issue. After a random (or so it seems) amount of steps, my script crashes with `msgpackrpc.error.TimeoutError: Request timed out` after calling `responses = self.simGetImages([ImageRequest(0, AirSimImageType.Scene, False, False), ImageRequest(0, AirSimImageType.DepthPerspective, True, False)])`. AirSim also closes during the process.

mallela, 2018-12-19:
I get the exact error as @paulfauthmayer. It happens approximately after ~2 days of training my DQN on the AirsimNH environment. It happens consistently. Unfortunately reproducing this takes a lot of time. 

@sytelus Is there somewhere I can access AirSim error log files to check what is happening and when exactly?

aoltonen, 2019-06-14:
@sytelus @Kjell-K Unfortunately I have a similar problem. The airsim crashed after a collision during a DQN training process under LandscapeMountains. And since that the airsim client could not be connected any more. 

Traceback (most recent call last):
 File "<ipython-input-1-1fb5dc1e49fe>", line 1, in <module>
    runfile('E:/Projects from Github/AirSim-master/PythonClient/multirotor/DQNdrone.py', wdir='E:/Projects from Github/AirSim-master/PythonClient/multirotor')
 File "D:\Anaconda3\envs\tensorflow\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 678, in runfile
    execfile(filename, namespace)
File "D:\Anaconda3\envs\tensorflow\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 106, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)
 File "E:/Projects from Github/AirSim-master/PythonClient/multirotor/DQNdrone.py", line 510, in <module>
    client.confirmConnection()
File "E:\Projects from Github\AirSim-master\PythonClient\airsim\client.py", line 54, in confirmConnection
    if self.ping():
  File "E:\Projects from Github\AirSim-master\PythonClient\airsim\client.py", line 24, in ping
    return self.client.call('ping')
  File "D:\Anaconda3\envs\tensorflow\lib\site-packages\msgpackrpc\session.py", line 41, in call
    return self.send_request(method, args).get()
  File "D:\Anaconda3\envs\tensorflow\lib\site-packages\msgpackrpc\future.py", line 43, in get
    raise self._error
TimeoutError: Request timed out

Wondering how to solve this...

patnolan33, 2019-12-13:
@aoltonen Have you figured out a fix or workaround to this issue? I am seemingly getting the same issue where after some random amount of time, a timeout occurs.

abhimanyuchadha96, 2020-08-01:
Unfortunately, the issue still persists. Random timeout exceptions take place.

gonzaloaguilarjimenez, 2020-11-05:
Same to me. Same error.

lknnnli, 2020-12-01:
same time out error here.

lknnnli, 2020-12-01:
> Unfortunately, the issue still persists. Random timeout exceptions take place.

hi，have you solved this issue now? this issue occurs to me recently

lknnnli, 2020-12-01:
> Same to me. Same error.

have you solved it? this also happens to me recently

gonzaloaguilarjimenez, 2020-12-01:
I didn't solve it. Actually I do not know where it comes from, but I switched from my eGPU to the Intel graphics card (working in Ubuntu 18.04) and it works (slowly for RL) but works. 

I guess it must be some issue between Nvidia drivers for Ubuntu and Unreal Engine, I do not think it comes from AirSim.

gonzaloaguilarjimenez, 2020-12-01:
The RL has already been working for 4 days and already done 160k iterations.

lknnnli, 2020-12-02:
> I didn't solve it. Actually I do not know where it comes from, but I switched from my eGPU to the Intel graphics card (working in Ubuntu 18.04) and it works (slowly for RL) but works.
>
> I guess it [...]

Hi, thank you for your warm reply, today what i did is i re-clone airsim from official github again and rebuilt, then it works. for me the issue is I think some old package is not compatible with the system.