microsoft/AirSim #666: Can't use Blocks env; fails to build libUE4Editor-AirSim.so

inic8, 2017-12-07:
hi,

I am following the steps here: https://github.com/Microsoft/AirSim/blob/master/docs/unreal_blocks.md

I am building it for Linux (ubuntu 16.04, UE 4.17.2)

**This step fails**: 4. If you get prompted for incompatible version and conversion, select In-place conversion which is usually under "More" options. If you get prompted for missing modules, make sure to select No so you don't exit.

**Compilation error is:**
In file included from ../../../AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/CameraDirector.h:4:
../../../AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/VehiclePawnWrapper.h:28:14: error: 
      cannot use defaulted default constructor of 'WrapperConfig' within
      'VehiclePawnWrapper' outside of member functions because 'is_fpv_vehicle'
      has an initializer
        bool is_fpv_vehicle = false; 
             ^
../../../AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/VehiclePawnWrapper.h:29:21: error: 
      cannot use defaulted default constructor of 'WrapperConfig' within
      'VehiclePawnWrapper' outside of member functions because
      'vehicle_config_name' has an initializer
        std::string vehicle_config_name = ""; //use the default config name
                    ^
../../../AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/VehiclePawnWrapper.h:30:14: error: 
      cannot use defaulted default constructor of 'WrapperConfig' within
      'VehiclePawnWrapper' outside of member functions because
      'enable_collisions' has an initializer
        bool enable_collisions = true; 
             ^
../../../AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/VehiclePawnWrapper.h:31:14: error: 
      cannot use defaulted default constructor of 'WrapperConfig' within
      'VehiclePawnWrapper' outside of member functions because
      'enable_passthrough_on_collisions' has an initializer
        bool enable_passthrough_on_collisions = false; 
             ^
../../../AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/VehiclePawnWrapper.h:32:15: error: 
      cannot use defaulted default constructor of 'WrapperConfig' within
      'VehiclePawnWrapper' outside of member functions because 'home_lattitude'
      has an initializer
        float home_lattitude = 47.641468;
              ^
../../../AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/VehiclePawnWrapper.h:33:15: error: 
      cannot use defaulted default constructor of 'WrapperConfig' within
      'VehiclePawnWrapper' outside of member functions because 'home_longitude'
      has an initializer
        float home_longitude = -122.140165;
              ^
../../../AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/VehiclePawnWrapper.h:34:15: error: 
      cannot use defaulted default constructor of 'WrapperConfig' within
      'VehiclePawnWrapper' outside of member functions because 'home_altitude'
      has an initializer
        float home_altitude = 122;
              ^
../../../AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/VehiclePawnWrapper.h:35:14: error: 
      cannot use defaulted default constructor of 'WrapperConfig' within
      'VehiclePawnWrapper' outside of member functions because 'enable_trace'
      has an initializer
        bool enable_trace = false;
             ^
../../../AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/VehiclePawnWrapper.h:42:105: note: 
      implicit default constructor for 'VehiclePawnWrapper::WrapperConfig' first
      required here
  ...std::vector<APIPCamera*>& cameras, const WrapperConfig& config = Wrapper...
                                                                      ^
In file included from /home/shashank/Code/Alpha/Projects/reinforcement-learning/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.cpp:14:
In file included from /home/shashank/Code/Alpha/Projects/reinforcement-learning/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirSimGameMode.cpp:4:
In file included from ../../../AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/SimHUD/SimHUD.h:6:
In file included from ../../../AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/SimMode/SimModeBase.h:5:
In file included from ../../../AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/CameraDirector.h:4:
../../../AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/VehiclePawnWrapper.h:28:14: error: 
      cannot use defaulted default constructor of 'WrapperConfig' within
[... 56 lines cut ...]
                    ^
../../../AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/VehiclePawnWrapper.h:104:19: note: 
      member is declared here
    WrapperConfig config_;
                  ^
../../../AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/VehiclePawnWrapper.h:27:12: note: 
      'VehiclePawnWrapper::WrapperConfig' declared here
    struct WrapperConfig {
           ^
8 errors generated.

**UE gui output is:**
...
Reflection code generated for BlocksEditor in 8.3575719 seconds
@progress pop
Using bundled libc++ standard C++ library.
Using fast way to relink  circularly dependent libraries (no FixDeps).
Performing 5 actions (20 in parallel)
[1/5] Compile Module.AirSim.cpp
[3/5] Compile Blocks.cpp
[2/5] Compile Module.AirSim.gen.cpp
ERROR: UBT ERROR: Failed to produce item: /home/shashank/Code/Alpha/Projects/reinforcement-learning/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Binaries/Linux/libUE4Editor-AirSim.so
Total build time: 108.22 seconds (Local executor: 0.00 seconds)

Thanks!

spsancti, 2017-12-07:
Facing the same issue now :(

Kalay1995, 2017-12-08:
in file VehiclePawnWrapper.h, change class VehiclePawnWrapper to 
'''c
class VehiclePawnWrapper
{
	public: //types
	typedef msr::airlib::GeoPoint GeoPoint;
	typedef msr::airlib::Vector3r Vector3r;
	typedef msr::airlib::Pose Pose;
	typedef msr::airlib::Quaternionr Quaternionr;
	typedef msr::airlib::CollisionInfo CollisionInfo;
	typedef msr::airlib::VectorMath VectorMath;
	typedef msr::airlib::real_T real_T;
	typedef msr::airlib::Utils Utils;
	
	public:
	struct WrapperConfig {
		bool is_fpv_vehicle;// = false; 
		std::string vehicle_config_name;// = ""; //use the default config name
		bool enable_collisions;// = true; 
		bool enable_passthrough_on_collisions;// = false; 
		float home_lattitude;// = 47.641468;
		float home_longitude;// = -122.140165;
		float home_altitude;// = 122;
		bool enable_trace;// = false;
	};
'''
and in file VehiclePawnWrapper.cpp, change VehiclePawnWrapper::VehiclePawnWrapper() to 
'''c
VehiclePawnWrapper::VehiclePawnWrapper()
{
	static ConstructorHelpers::FObjectFinder<UParticleSystem> collision_display(TEXT("ParticleSystem'/AirSim/StarterContent/Particles/P_Explosion.P_Explosion'"));
	if (!collision_display.Succeeded())
	collision_display_template = collision_display.Object;
	else
	collision_display_template = nullptr;
	
	config_.is_fpv_vehicle = false; 
	config_.vehicle_config_name = ""; //use the default config name
	config_.enable_collisions = true; 
	config_.enable_passthrough_on_collisions = false; 
	config_.home_lattitude = 47.641468;
	config_.home_longitude = -122.140165;
	config_.home_altitude = 122;
	config_.enable_trace = false;
}
'''

Kalay1995, 2017-12-08:
But I face another problem: 
I can spawn car and control it normally. But when I spawn quadrotor, the scene would be white, unfortunately.

inic8, 2017-12-08:
@spsancti What @Kalay1995  proposes regarding braces-or-equal initializers for nested classes solves the issue as far as car simulation is concerned. Thanks @Kalay1995.

closed 2017-12-08 by inic8

reinaldomaslim, 2017-12-11:
@Kalay1995 did I need to re-setup and rebuild for airsim after the change? Thanks

sytelus, 2017-12-12:
@Kalay1995 - your proposed fix is bit problematic. I've added proper fix in master branch.