microsoft/AirSim #668: use Blocks env,  when play and select quadrotor, I got a white scene.

Kalay1995, 2017-12-08:
Platform: Ubuntu 16.04 LTS, UE4.17.2

I follow the setps [https://github.com/Microsoft/AirSim/blob/master/docs/unreal_blocks.md](url), and fixed problem #666 . When I get into the Blocks environment, and play it using quadrotor simulation, I got a white scene:
[image]

But at the meantime, if I select car simulation, It works well:
[image]
also I can control the car use the keyboard.

In addition, I got some warning in the terminal:
It looks like:
[2017.12.08-03.57.15:461][196]LogActor: Warning: BP_CameraDirector_C /Game/FlyingCPP/Maps/UEDPIE_0_FlyingExampleMap.FlyingExampleMap:PersistentLevel.BP_CameraDirector_C_0 has natively added scene component(s), but none of them were set as the actor's RootComponent - picking one arbitrarily
[2017.12.08-03.57.15:466][196]PIE: Play in editor start time for /Game/FlyingCPP/Maps/UEDPIE_0_FlyingExampleMap 2.62
[2017.12.08-03.57.15:467][196]LogBlueprintUserMessages: Late PlayInEditor Detection: Level '/Game/FlyingCPP/Maps/FlyingExampleMap.FlyingExampleMap:PersistentLevel' has LevelScriptBlueprint '/Game/FlyingCPP/Maps/FlyingExampleMap.FlyingExampleMap:PersistentLevel.FlyingExampleMap' with GeneratedClass '/Game/FlyingCPP/Maps/FlyingExampleMap.FlyingExampleMap_C' with ClassGeneratedBy '/Game/FlyingCPP/Maps/FlyingExampleMap.FlyingExampleMap:PersistentLevel.FlyingExampleMap'
[2017.12.08-03.57.50:287][372]LogHttp: Warning: 0x7ff3b4110280: request failed, libcurl error: 60 (Peer certificate cannot be authenticated with given CA certificates)
[2017.12.08-03.57.50:287][372]LogHttp: Warning: 0x7ff3b4110280: libcurl info message cache 0 (Hostname in DNS cache was stale, zapped)
[2017.12.08-03.57.50:287][372]LogHttp: Warning: 0x7ff3b4110280: libcurl info message cache 1 (  Trying 119.28.34.46...)
[2017.12.08-03.57.50:287][372]LogHttp: Warning: 0x7ff3b4110280: libcurl info message cache 2 (Connected to datarouter.ol.epicgames.com (119.28.34.46) port 443 (#26))
[2017.12.08-03.57.50:287][372]LogHttp: Warning: 0x7ff3b4110280: libcurl info message cache 3 (Cipher selection: ALL:!EXPORT:!EXPORT40:!EXPORT56:!aNULL:!LOW:!RC4:@STRENGTH)
[2017.12.08-03.57.50:287][372]LogHttp: Warning: 0x7ff3b4110280: libcurl info message cache 4 (successfully set certificate verify locations:)
[2017.12.08-03.57.50:287][372]LogHttp: Warning: 0x7ff3b4110280: libcurl info message cache 5 (  CAfile: /etc/ssl/certs/ca-certificates.crt  CApath: none)
[2017.12.08-03.57.50:287][372]LogHttp: Warning: 0x7ff3b4110280: libcurl info message cache 6 (TLSv1.2 (OUT), TLS Unknown, Certificate Status (22):)
[2017.12.08-03.57.50:287][372]LogHttp: Warning: 0x7ff3b4110280: libcurl info message cache 7 (TLSv1.2 (OUT), TLS handshake, Client hello (1):)
[2017.12.08-03.57.50:287][372]LogHttp: Warning: 0x7ff3b4110280: libcurl info message cache 8 (TLSv1.2 (IN), TLS handshake, Server hello (2):)
[2017.12.08-03.57.50:287][372]LogHttp: Warning: 0x7ff3b4110280: libcurl info message cache 9 (NPN, negotiated HTTP1.1)
[2017.12.08-03.57.50:287][372]LogHttp: Warning: 0x7ff3b4110280: libcurl info message cache 10 (TLSv1.2 (IN), TLS handshake, Certificate (11):)
[2017.12.08-03.57.50:287][372]LogHttp: Warning: 0x7ff3b4110280: libcurl info message cache 11 (TLSv1.2 (OUT), TLS alert, Server hello (2):)
[2017.12.08-03.57.50:287][372]LogHttp: Warning: 0x7ff3b4110280: libcurl info message cache 12 (SSL certificate problem: unable to get local issuer certificate)
[2017.12.08-03.57.50:287][372]LogHttp: Warning: 0x7ff3b4110280: libcurl info message cache 13 (Closing connection 26)
[2017.12.08-03.57.50:288][372]LogHttp: Warning: 0x7ff3b4110280: libcurl info message cache 14 (TLSv1.2 (OUT), TLS alert, Client hello (1):)

and Warning:
[2017.12.08-03.55.34:476][717]LogActor: Warning: BP_CameraDirector_C /Game/FlyingCPP/Maps/UEDPIE_0_FlyingExampleMap.FlyingExampleMap:PersistentLevel.BP_CameraDirector_C_0 has natively added scene component(s), but none of them were set as the actor's RootComponent - picking one arbitrarily

Thanks.

satrialoka, 2017-12-15:
it seems that the quadrotor is going down trough the ground, i tried to press the backspace button to see what happen.
have you solve this issue?

Kalay1995, 2017-12-15:
No, I have not solved it. I agree with you, but after some attempts, I
failed to fix it and turned to do other things. Hope you can solve it.

sytelus, 2017-12-16:
I got latest version, compiled and ran on Ubuntu 16.04 without any issues. Quad as sitting on floor as it should. Did you modified anything? Does this happen when you just press Play button?

Kalay1995, 2017-12-16:
I think I should get the latest version first, thank you.

closed 2018-01-13 by sytelus