microsoft/AirSim #669: BSP not getting segmentation

oppne, 2017-12-08:
Hey! Was just wondering, why isnt bsp meshes colorized?

sytelus, 2017-12-11:
Currently we iterate all objects of type UMeshComponent and assign then custom stencil ID value from 0 to 255, each of which translates to a color. Any object that is not this type does not get assigned custom stencil ID and hence won't have segmentation color assigned.

I would suggest to play with this code which does above to include object types of your interest. Please feel free to submit a pull request or change suggestion that works for objects you are interested in.
https://github.com/Microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/AirBlueprintLib.cpp#L183

oppne, 2017-12-12:
Well, i have already solved this in blueprints! But this is converting a 8 bit to 32 bit and setting the color using a few nodes in the material that lerps the color and replaces it.

sytelus, 2018-02-13:
Could you please post screenshot of your blueprint? It would be nice to add this in AirSim.

closed 2019-09-27 by ironclownfish