microsoft/AirSim #671: in which frame is reportState reporting?

abdelrahman-t, 2017-12-09:
reporter.writeValue("Position", current_.pose.position);
        reporter.writeValue("Orientation", current_.pose.orientation);
        reporter.writeValue("Lin-Vel", current_.twist.linear);
        reporter.writeValue("Lin-Accl", current_.accelerations.linear);
        reporter.writeValue("Ang-Vel", current_.twist.angular);
        reporter.writeValue("Ang-Accl", current_.accelerations.angular);

could you please verify which frame are these reported values in?

sytelus, 2017-12-12:
Linear quantities are in world frame while the angular quantities are in body frame. The coordinate system is NED, all units are SI.

abdelrahman-t, 2017-12-12:
@sytelus thank you!

closed 2017-12-12 by abdelrahman-t