microsoft/AirSim #672: Is there a method to find out about drone's energy consumption?

moeenmh, 2017-12-09:
Hello there, I would like to know if its possible to check the energy consumption of the drone? (after it covers a certain path, let's say)

Thank you in advance

saihv, 2017-12-09:
Just a very high level idea: I believe the drones in AirSim are modeled with the MT2212 motors and the GWS 9X5 props in mind. There should also be a thrust calculation somewhere in the code. You should be able to relate the commanded throttle to the thrust, and subsequently to motor current through the thrust vs throttle and current vs thrust curves from the motor datasheets. Of course, this would be approximate and only for the motors.

sytelus, 2017-12-12:
I would suggest to look in to [this introduction, page 3](https://www.researchgate.net/file.PostFileLoader.html?id=57795c0ff7b67efa2d7d0171&assetKey=AS%3A379835361841153%401467571215713). The power consumed by a rotor is given by (torque * angular velocity * constant). These quantities are available in [this function](https://github.com/Microsoft/AirSim/blob/master/AirLib/include/vehicles/multirotor/Rotor.hpp#L115) so you can easily compute power consumed per rotor.  Then in [this function](https://github.com/Microsoft/AirSim/blob/master/AirLib/include/vehicles/multirotor/MultiRotor.hpp#L75) you can sum total power consumed by all rotors + add some fixed amount for onboard electronics.

RVBldr, 2018-11-21:
Internal VSTF 7415

kaioforte, 2019-07-02:
Hi, I was trying to calculate battery consumption of the drone, following the tip given by @sytelus . I could not find a way to access in code the info required to the calculations (torque, angular velocity) present in this [file](https://github.com/Microsoft/AirSim/blob/master/AirLib/include/vehicles/multirotor/Rotor.hpp#L115) using the API or any other method.

Do you have a solution for this problem?

jonyMarino, 2020-10-19:
Closed due to inactivity. Doc update pending

closed 2020-10-19 by jonyMarino