microsoft/AirSim #673: In-place conversion under more option is failed, can not be compiled

yunxiaoshan, 2017-12-10:
system :ubuntu 16.04

`The project could not be compiled. Would you like to open it in an IDE to edit source code?

Running /home/shanyx/Downloads/UnrealEngine/Engine/Binaries/DotNET/UnrealBuildTool.exe Blocks Development Linux -project="/home/shanyx/Downloads/AirSim/Unreal/Environments/Blocks/Blocks.uproject" -editorrecompile -progress -NoHotReloadFromIDE
Setting swap interval to 'Immediate'

Running Mono...

~/Downloads/UnrealEngine/Engine ~/Downloads/UnrealEngine/Engine/Binaries/Linux
Creating makefile for Blocks (no existing makefile)
Performing full C++ include scan (no include cache file)
Using clang version '3.9.1' (string), 3 (major), 9 (minor), 1 (patch)
@progress push 5%
Building UnrealHeaderTool...
Creating makefile for UnrealHeaderTool (no existing makefile)
Using clang version '3.9.1' (string), 3 (major), 9 (minor), 1 (patch)
Using bundled libc++ standard C++ library.
Using fast way to relink  circularly dependent libraries (no FixDeps).
Target is up to date
Total build time: 1.48 seconds (NoActionsToExecute executor: 0.00 seconds)
Parsing headers for BlocksEditor
  Running UnrealHeaderTool "/home/shanyx/Downloads/AirSim/Unreal/Environments/Blocks/Blocks.uproject" "/home/shanyx/Downloads/AirSim/Unreal/Environments/Blocks/Intermediate/Build/Linux/B4D820EA/BlocksEditor/Development/BlocksEditor.uhtmanifest" -LogCmds="loginit warning, logexit warning, logdatabase error" -Unattended -WarningsAsErrors
LogPlatformFile: Not using cached read wrapper
LogInit: Display: RandInit(-1916358273) SRandInit(-1916358273).
LogTaskGraph: Started task graph with 4 named threads and 5 total threads with 1 sets of task threads.
This project requires the 'UdpMessaging' plugin, which is not compatible with the current engine version.
Would you like to disable it? You will no longer be able to open any assets created using it.
LogPluginManager: Mounting plugin LevelSequenceEditor
LogPluginManager: Mounting plugin MatineeToLevelSequence
LogPluginManager: Mounting plugin Paper2D
LogPluginManager: Mounting plugin PhysXVehicles
LogPluginManager: Mounting plugin ImmediatePhysics
LogPluginManager: Mounting plugin GoogleCloudMessaging
LogPluginManager: Mounting plugin SteamVR
LogPluginManager: Mounting plugin ProceduralMeshComponent
LogPluginManager: Mounting plugin LocationServicesBPLibrary
LogPluginManager: Mounting plugin OculusVR
LogPluginManager: Mounting plugin RuntimePhysXCooking
LogPluginManager: Mounting plugin CableComponent
LogPluginManager: Mounting plugin IOSDeviceProfileSelector
LogPluginManager: Mounting plugin AndroidDeviceProfileSelector
LogPluginManager: Mounting plugin MobilePatchingUtils
LogPluginManager: Mounting plugin AndroidMoviePlayer
LogPluginManager: Mounting plugin AppleMoviePlayer
LogPluginManager: Mounting plugin ExampleDeviceProfileSelector
LogPluginManager: Mounting plugin ArchVisCharacter
LogPluginManager: Mounting plugin AndroidPermission
LogPluginManager: Mounting plugin CustomMeshComponent
LogPluginManager: Mounting plugin WindowsMoviePlayer
LogPluginManager: Mounting plugin LinuxDeviceProfileSelector
LogPluginManager: Mounting plugin OnlineSubsystemNull
LogPluginManager: Mounting plugin OnlineSubsystemUtils
LogPluginManager: Mounting plugin OnlineSubsystem
LogPluginManager: Mounting plugin OnlineSubsystemIOS
LogPluginManager: Mounting plugin OnlineSubsystemGooglePlay
LogPluginManager: Mounting plugin UObjectPlugin
LogPluginManager: Mounting plugin NullSourceCodeAccess
LogPluginManager: Mounting plugin XCodeSourceCodeAccess
LogPluginManager: Mounting plugin PerforceSourceControl
LogPluginManager: Mounting plugin VisualStudioSourceCodeAccess
LogPluginManager: Mounting plugin GitSourceControl
LogPluginManager: Mounting plugin KDevelopSourceCodeAccess
LogPluginManager: Mounting plugin CodeLiteSourceCodeAccess
LogPluginManager: Mounting plugin SubversionSourceControl
LogPluginManager: Mounting plugin TcpMessaging
LogPluginManager: Mounting plugin UdpMessaging
LogPluginManager: Mounting plugin AndroidMedia
LogPluginManager: Mounting plugin WmfMedia
LogPluginManager: Mounting plugin MediaPlayerEditor
[... 1609 lines cut ...]
[2017.12.10-01.44.16:288][  0]LogCompile: Autogenerating boilerplate cpp: AudioEditor.gen.cpp
[2017.12.10-01.44.16:293][  0]LogCompile: Success: Parsed 1154 line(s), 404 statement(s) in 0.01 secs.
[2017.12.10-01.44.16:321][  0]LogCompile: Autogenerating boilerplate cpp: ViewportInteraction.gen.cpp
[2017.12.10-01.44.16:326][  0]LogCompile: Success: Parsed 2119 line(s), 657 statement(s) in 0.01 secs.
[2017.12.10-01.44.16:359][  0]LogCompile: Autogenerating boilerplate cpp: VREditor.gen.cpp
[2017.12.10-01.44.16:364][  0]LogCompile: Success: Parsed 3013 line(s), 971 statement(s) in 0.01 secs.
[2017.12.10-01.44.16:385][  0]LogCompile: Autogenerating boilerplate cpp: AnimationModifiers.gen.cpp
[2017.12.10-01.44.16:391][  0]LogCompile: Success: Parsed 505 line(s), 160 statement(s) in 0.02 secs.
[2017.12.10-01.44.16:395][  0]LogCompile: Autogenerating boilerplate cpp: SequenceRecorderSections.gen.cpp
[2017.12.10-01.44.16:398][  0]LogCompile: Success: Parsed 98 line(s), 30 statement(s) in 0.00 secs.
[2017.12.10-01.44.16:412][  0]LogCompile: Autogenerating boilerplate cpp: UMGEditor.gen.cpp
[2017.12.10-01.44.16:416][  0]LogCompile: Success: Parsed 584 line(s), 192 statement(s) in 0.01 secs.
[2017.12.10-01.44.16:431][  0]LogCompile: Autogenerating boilerplate cpp: LandscapeEditor.gen.cpp
[2017.12.10-01.44.16:435][  0]LogCompile: Success: Parsed 911 line(s), 178 statement(s) in 0.01 secs.
[2017.12.10-01.44.16:441][  0]LogCompile: Autogenerating boilerplate cpp: FoliageEdit.gen.cpp
[2017.12.10-01.44.16:444][  0]LogCompile: Success: Parsed 136 line(s), 62 statement(s) in 0.00 secs.
[2017.12.10-01.44.16:449][  0]LogCompile: Autogenerating boilerplate cpp: WorldBrowser.gen.cpp
[2017.12.10-01.44.16:452][  0]LogCompile: Success: Parsed 274 line(s), 58 statement(s) in 0.00 secs.
[2017.12.10-01.44.16:467][  0]LogCompile: Autogenerating boilerplate cpp: IntroTutorials.gen.cpp
[2017.12.10-01.44.16:471][  0]LogCompile: Success: Parsed 599 line(s), 137 statement(s) in 0.01 secs.
[2017.12.10-01.44.16:472][  0]LogCompile: Autogenerating boilerplate cpp: LocalizationDashboard.gen.cpp
[2017.12.10-01.44.16:475][  0]LogCompile: Success: Parsed 16 line(s), 8 statement(s) in 0.00 secs.
[2017.12.10-01.44.16:477][  0]LogCompile: Autogenerating boilerplate cpp: DestructibleMeshEditor.gen.cpp
[2017.12.10-01.44.16:481][  0]LogCompile: Success: Parsed 36 line(s), 18 statement(s) in 0.00 secs.
[2017.12.10-01.44.16:484][  0]LogCompile

kecaiwu, 2017-12-11:
same issues!

sibojia, 2017-12-11:
same issues. More info on the compile error:
```
In file included from /home/msim/work/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.gen.cpp:5:
In file included from /home/msim/work/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Inc/AirSim/CameraDirector.gen.cpp:8:
In file included from ../../../AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/CameraDirector.h:4:
../../../AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/VehiclePawnWrapper.h:28:14: error: cannot
      use defaulted default constructor of 'WrapperConfig' within 'VehiclePawnWrapper' outside of member
      functions because 'is_fpv_vehicle' has an initializer
        bool is_fpv_vehicle = false; 
             ^
../../../AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/VehiclePawnWrapper.h:29:21: error: cannot
      use defaulted default constructor of 'WrapperConfig' within 'VehiclePawnWrapper' outside of member
      functions because 'vehicle_config_name' has an initializer
        std::string vehicle_config_name = ""; //use the default config name
                    ^
../../../AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/VehiclePawnWrapper.h:30:14: error: cannot
      use defaulted default constructor of 'WrapperConfig' within 'VehiclePawnWrapper' outside of member
      functions because 'enable_collisions' has an initializer
        bool enable_collisions = true; 
             ^
../../../AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/VehiclePawnWrapper.h:31:14: error: cannot
      use defaulted default constructor of 'WrapperConfig' within 'VehiclePawnWrapper' outside of member
      functions because 'enable_passthrough_on_collisions' has an initializer
        bool enable_passthrough_on_collisions = false; 
             ^
../../../AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/VehiclePawnWrapper.h:32:15: error: cannot
      use defaulted default constructor of 'WrapperConfig' within 'VehiclePawnWrapper' outside of member
      functions because 'home_lattitude' has an initializer
        float home_lattitude = 47.641468;
              ^
../../../AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/VehiclePawnWrapper.h:33:15: error: cannot
      use defaulted default constructor of 'WrapperConfig' within 'VehiclePawnWrapper' outside of member
      functions because 'home_longitude' has an initializer
        float home_longitude = -122.140165;
              ^
../../../AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/VehiclePawnWrapper.h:34:15: error: cannot
      use defaulted default constructor of 'WrapperConfig' within 'VehiclePawnWrapper' outside of member
      functions because 'home_altitude' has an initializer
        float home_altitude = 122;
              ^
../../../AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/VehiclePawnWrapper.h:35:14: error: cannot
      use defaulted default constructor of 'WrapperConfig' within 'VehiclePawnWrapper' outside of member
      functions because 'enable_trace' has an initializer
        bool enable_trace = false;
../../../AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/VehiclePawnWrapper.h:42:105: note: 
      implicit default constructor for 'VehiclePawnWrapper::WrapperConfig' first required here
  ...pawn, const std::vector<APIPCamera*>& cameras, const WrapperConfig& config = WrapperConfig());
                                                                                  ^
In file included from /home/msim/work/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.cpp:37:
/home/msim/work/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/VehiclePawnWrapper.cpp:11:21: error: 
      constructor for 'VehiclePawnWrapper' must explicitly initialize the member 'config_' which does not
      have a default constructor
VehiclePawnWrapper::VehiclePawnWrapper()
                    ^
../../../AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/VehiclePawnWrapper.h:104:19: note: member
      is declared here
    WrapperConfig config_;
                  ^
../../../AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/VehiclePawnWrapper.h:27:12: note: 
      'VehiclePawnWrapper::WrapperConfig' declared here
    struct WrapperConfig {
           ^
8 errors generated.

```

sibojia, 2017-12-11:
I managed to solve the issue with the method in #653 . Seems like a compiler issue of `cannot use defaulted default constructor of 'WrapperConfig' within 'VehiclePawnWrapper' outside of member
      functions because 'xxx' has an initializer`. However I don't have a newer clang installed as in #653. I'm compiling AirSim on a clean Ubuntu 16.04 install.

sytelus, 2017-12-12:
This is duplicate of https://github.com/Microsoft/AirSim/issues/666. This issue is now fixed in master and tested on Ubuntu 16.04.

closed 2017-12-12 by sytelus