microsoft/AirSim #675: Reset world level on reset event

deanzadok, 2017-12-10:
Hi everyone,

I want to be able to reset the world level whenever I call the reset method from the python client.
by reseting the world, I mean reseting every objects locations, so if I hit an object with my car and call reset, I want to be able to restore the object's original location.

we are working with the suv, so we tried this c++ command under one of the implementations of reset in vehicleapibase and vehiclepawnwrapper:
http://www.unrealidiot.com/how-to-reload-current-level-in-unreal-engine-4
but it didn't work out since the vehicle is not an implementation of an actor.

Any suggestions to solve this in a simple way?

Thanks in advance.

sytelus, 2017-12-11:
You can access the `pawn_` variable in VehiclePawnWrapper which is instance of an actor.  However be careful with this because this will unload all the current objects so unless this is done carefully you might get a crash.

deanzadok, 2017-12-11:
Hi Shital,

Thanks for the comment!
we managed to compile the command under the reset event in VehiclePawnWrapper.cpp:
UGameplayStatics::OpenLevel(pawn_->GetActorClass(), FName(FName(*((pawn_->GetWorld())->GetName())), false));

So we get there when we're on reset event but nothing happened. any idea what is our problem?
is this the right way to solve this?

Thanks.

sytelus, 2018-02-13:
I'm thinking we should add new API called resetWorld which will reload level. If you this working, you might want to send pull request :).

srivatsankrishnan, 2018-10-30:
@sytelus @deanzadok 
Is this API implemented? I am also looking for this functionality and it would be great if this is already implemented.

karnikram, 2021-02-09:
+1 for this API request. It's needed especially when there are many dynamic characters in the scene.

ugurkanates, 2021-05-24:
+1 as well. We are running blueprints for dynamic objects. Would be great to have something native