microsoft/AirSim #679: how can I use Windows SOCKET TCP in AirSim

abc19850508, 2017-12-13:
I am trying to send images to another computer in AirSim. And I have written the code sending images with Windows SOCKET. When I tried to move the code into AirSim,  I found that a lot of methods and classes were redefined. So I want to ask how should I solve this problem if I want to use Windows SOCKET. Thanks a lot for your answers!

sytelus, 2017-12-16:
Not sure what you mean. Please provide more details on what exactly you are doing, copy paste error messages etc. If it is custom code and if you don't have background in C++ development then you might also want to seek help from nearest expert around you. Unfortunately we do not have resources to debug custom code.

abc19850508, 2017-12-18:
@sytelus Sorry! I didn't describe the problem clearly! I try again. 
My purpose is to transfer images from AirSim to another computer with TCP. 
I have debugged the code with console program in visual studio 2015. 
When I want to move the code into AirSim project. I found that a lot of api which were used in my former code were redefined, you can see the errors in the following picture. Sorry I can't change my visual studio 2015 to English version, there is always internet problem when I want to download something from Microsoft. The error is redefinition.
I think the problem is that UE4 redefine TCP APIs.  I thought of two options. First, I insist on using tcp API in <winsock2.h> <ws2tcpip.h>. That is more easily for me. I have debugged the code. But I don't know how to solve the redefinition errors. That is what I ask in this issue. Second, use the API which is defined by UE4. But I can't find enough examples. That is really hard for me to write code just with the help of UE4 document. I am a beginner in C++. But I am trying the second way now.There is no c
++ expert around me. If it is possible can you give me some advice on how to solve the redefinition error or provide some UE4 TCP examples. Thanks a lot. 
[image]

abc19850508, 2017-12-19:
In a nutshell， how to use Windows Socket instead of FSocket in AirSim, is it possible? @sytelus

sytelus, 2017-12-19:
If I understand correctly, your are actually introducing new files in AirSim that uses direct WIndows APIs. Unreal doesn't support including direct Windows headers very well and usual recommended way is to use Unreal APIs instead. If you really want to use Windows APIs in Unreal code then there are ways but it would require you to research on places like https://answers.unrealengine.com. For example see [this post](https://answers.unrealengine.com/questions/499594/windowsh-error-in-new-c-project.html). AirSim uses some Windows APIs directly but it was hard to get it work. You can search for windows.h in AirSim code for examples.

lovettchris, 2017-12-21:
Perhaps all you need to do is build your socket code into a separate library like we have already done in MavLinkCom library (which has sockets code inside already), in fact, it has a cross-platform socket class called TcpClientPort that you might be able to just reuse as is.  It also has UdpClientPort.  Udp can be a lot faster, if you don't care about possibly losing a few packets.

abc19850508, 2017-12-22:
Thanks a lot for your advice! @sytelus @lovettchris  It is really a bad experience to use Windows APIs in UE4. One more thing, if I want to add some code in RecordingThread.cpp(the path is C:\Users\Windows\Documents\Unreal Projects\LandscapeMountains\Plugins\AirSim\Source\Recording\RecordingThread.cpp in LandscapeMountains). Do I need to add the code to RecordingThread.cpp in AirSim project first, and then rebuild it, and move the Plugins folder into LandscapeMountains or I can add the code directly into the LandscapeMountains' RecordingThread.cpp

sytelus, 2017-12-29:
You can just change files in your Unreal project. However if you decide to sync with latest version then you will have to merge your changes (instead of just copieng over Plugins folder from AirSim because that will overwrite your changes).

closed 2018-02-13 by sytelus