microsoft/AirSim #680: can not include any files of AirSim Plugins

Jackyjae, 2017-12-13:
I built the AirSim successfully but when I copy the plugins into my own project, I got this errors from the VS.
[image]
And when I compile it at UE editor, I got this.
[image]
Is it about the problems of "AirSim.Build.cs" or "Myproject.Build.cs"
I don't know how to fix it and I can't use the AirSim's APIs, NEED HELPS.

sytelus, 2017-12-16:
Did you copied Plugins folder after the build? Then run clean.bat and GenerateSolutionFiles.bat from your project?

Please list the steps you followed. Also, copy paste from Output window instead of screenshots.

Jackyjae, 2017-12-16:
I built the AirSim with vs2017 and copied the plugin folder into my project path but I can't find any clean.bat and GenerateSolutionFiles.bat files under my project folder. For now, the errors are:
`Severity	Code	Description	Project	File	Line	Suppression State
Error	MSB3075	The command ""D:\Epic Games\UE_4.18\Engine\Build\BatchFiles\Build.bat" Task1Editor Win64 DebugGame "C:\Users\angus\Documents\Unreal Projects\Task1\Task1.uproject" -waitmutex" exited with code 5. Please verify that you have sufficient rights to run this command.	Task1	C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.MakeFile.Targets	41	

Severity	Code	Description	Project	File	Line	Suppression State
Error		Failed to produce item: C:\Users\angus\Documents\Unreal Projects\Task1\Binaries\Win64\UE4Editor-Task1-Win64-DebugGame.pdb	Task1	C:\Users\angus\Documents\Unreal Projects\Task1\Intermediate\ProjectFiles\ERROR	1	

Severity	Code	Description	Project	File	Line	Suppression State
Error	C1083	Cannot open include file: 'vehicles/multirotor/api/MultirotorRpcLibClient.hpp': No such file or directory	Task1	c:\users\angus\documents\unreal projects\task1\source\task1\SocketServer.h	3	
`
Thank you.

Jackyjae, 2017-12-16:
@sytelus BTW, my Task.uproject file is like this:

{
	"FileVersion": 3,
	"EngineAssociation": "4.18",
	"Category": "",
	"Description": "",
	"Modules": [
		{
			"Name": "Task1",
			"Type": "Runtime",
			"LoadingPhase": "Default",
			"AdditionalDependencies": [
 "AirSim"
]
		}
	],
	"Plugins": [
 {
             "Name": "AirSim",
             "Enabled": true
         },
		{
			"Name": "PhysXVehicles",
			"Enabled": true
		},
		{
			"Name": "RawInput",
			"Enabled": true
		},
		{
			"Name": "OpenCV",
			"Enabled": true
		}
	]
}

Jackyjae, 2017-12-16:
1>------ Build started: Project: Task1, Configuration: DebugGame_Editor x64 ------
1>D:\IDE\VS\Common7\IDE\VC\VCTargets\Microsoft.Cpp.props(31,3): warning MSB4011: "D:\IDE\VS\Common7\IDE\VC\VCTargets\Microsoft.Makefile.props" cannot be imported again. It was already imported at "C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.cpp.props (31,3)". This is most likely a build authoring error. This subsequent import will be ignored. [C:\Users\angus\Documents\Unreal Projects\Task1\Intermediate\ProjectFiles\Task1.vcxproj]
1>EXEC : warning : C:\Users\angus\Documents\Unreal Projects\Task1\Plugins\OpenCV\Source\OpenCV\OpenCV.Build.cs: warning: Module constructors should take a ReadOnlyTargetRules argument (rather than a TargetInfo argument) and pass it to the base class constructor from 4.15 onwards. Please update the method signature.
1>Compiling game modules for hot reload
1>Performing 3 actions (4 in parallel)
1>[1/3] Link UE4Editor-AirSim-3020-Win64-DebugGame.dll
1>[2/3] Link UE4Editor-AirSim-3020-Win64-DebugGame.lib
1>   Creating library C:\Users\angus\Documents\Unreal Projects\Task1\Plugins\AirSim\Intermediate\Build\Win64\UE4Editor\DebugGame\UE4Editor-AirSim-3020-Win64-DebugGame.lib and object C:\Users\angus\Documents\Unreal Projects\Task1\Plugins\AirSim\Intermediate\Build\Win64\UE4Editor\DebugGame\UE4Editor-AirSim-3020-Win64-DebugGame.exp
1>[3/3] Link UE4Editor-Task1-2620-Win64-DebugGame.dll
1>   Creating library C:\Users\angus\Documents\Unreal Projects\Task1\Plugins\AirSim\Intermediate\Build\Win64\UE4Editor\DebugGame\UE4Editor-AirSim-3020-Win64-DebugGame.suppressed.lib and object C:\Users\angus\Documents\Unreal Projects\Task1\Plugins\AirSim\Intermediate\Build\Win64\UE4Editor\DebugGame\UE4Editor-AirSim-3020-Win64-DebugGame.suppressed.exp
1>AirLib.lib(RpcLibClientBase.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '2' doesn't match value '0' in SharedPCH.Engine.NonOptimized.h.obj
1>AirLib.lib(RpcLibClientBase.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MDd_DynamicDebug' doesn't match value 'MD_DynamicRelease' in SharedPCH.Engine.NonOptimized.h.obj
1>AirLib.lib(RpcLibClientBase.obj) : warning LNK4075: ignoring '/EDITANDCONTINUE' due to '/OPT:LBR' specification
1>AirLib.lib(MultirotorRpcLibClient.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '2' doesn't match value '0' in SharedPCH.Engine.NonOptimized.h.obj
1>AirLib.lib(MultirotorRpcLibClient.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MDd_DynamicDebug' doesn't match value 'MD_DynamicRelease' in SharedPCH.Engine.NonOptimized.h.obj
1>AirLib.lib(client.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '2' doesn't match value '0' in SharedPCH.Engine.NonOptimized.h.obj
1>AirLib.lib(client.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MDd_DynamicDebug' doesn't match value 'MD_DynamicRelease' in SharedPCH.Engine.NonOptimized.h.obj
1>AirLib.lib(rpc_error.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '2' doesn't match value '0' in SharedPCH.Engine.NonOptimized.h.obj
1>AirLib.lib(rpc_error.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MDd_DynamicDebug' doesn't match value 'MD_DynamicRelease' in SharedPCH.Engine.NonOptimized.h.obj
1>AirLib.lib(response.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '2' doesn't match value '0' in SharedPCH.Engine.NonOptimized.h.obj
1>AirLib.lib(response.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MDd_DynamicDebug' doesn't match value 'MD_DynamicRelease' in SharedPCH.Engine.NonOptimized.h.obj
1>AirLib.lib(format.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '2' doesn't match value '0' in SharedPCH.Engine.NonOptimized.h.obj
1>AirLib.lib(format.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MDd_DynamicDebug' doesn't match value 'MD_DynamicRelease' in SharedPCH.Engine.NonOptimized.h.obj
1>   Creating library C:\Users\angus\Documents\Unreal Projects\Task1\Intermediate\Build\Win64\UE4Editor\DebugGame\UE4Editor-Task1-2620-Win64-DebugGame.suppressed.lib and object C:\Users\angus\Documents\Unreal Projects\Task1\Intermediate\Build\Win64\UE4Editor\DebugGame\UE4Editor-Task1-2620-Win64-DebugGame.suppressed.exp
1>AirLib.lib(format.obj) : error LNK2001: unresolved external symbol __imp__invalid_parameter
1>AirLib.lib(RpcLibClientBase.obj) : error LNK2001: unresolved external symbol __imp__invalid_parameter
1>AirLib.lib(MultirotorRpcLibClient.obj) : error LNK2001: unresolved external symbol __imp__invalid_parameter
1>AirLib.lib(client.obj) : error LNK2001: unresolved external symbol __imp__invalid_parameter
1>AirLib.lib(rpc_error.obj) : error LNK2001: unresolved external symbol __imp__invalid_parameter
1>AirLib.lib(format.obj) : error LNK2001: unresolved external symbol __imp__CrtDbgReportW
1>AirLib.lib(RpcLibClientBase.obj) : error LNK2001: unresolved external symbol __imp__CrtDbgReportW
1>AirLib.lib(MultirotorRpcLibClient.obj) : error LNK2001: unresolved external symbol __imp__CrtDbgReportW
1>AirLib.lib(client.obj) : error LNK2001: unresolved external symbol __imp__CrtDbgReportW
1>AirLib.lib(rpc_error.obj) : error LNK2001: unresolved external symbol __imp__CrtDbgReportW
[... cut ...]
1>Total build time: 7.83 seconds (Local executor: 0.00 seconds)
1>C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.MakeFile.Targets(41,5): error MSB3075: The command ""D:\Epic Games\UE_4.18\Engine\Build\BatchFiles\Build.bat" Task1Editor Win64 DebugGame "C:\Users\angus\Documents\Unreal Projects\Task1\Task1.uproject" -waitmutex" exited with code 5. Please verify that you have sufficient rights to run this command.
1>Done building project "Task1.vcxproj" -- FAILED.
========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========

sytelus, 2018-02-13:
VS2017 isn't supported currently. I would suggest to follow the steps as specified in our docs.

closed 2018-02-13 by sytelus