microsoft/AirSim #681: AttributeError: 'numpy.int32' object has no attribute 'to_msgpack'

lo2aayy, 2017-12-13:
Hey,

I am using computer vision mode to collect data using AirSim and I'm getting the following error:
" AttributeError: 'numpy.int32' object has no attribute 'to_msgpack' "

which is triggered by the following line:
'  client.simSetPose(Pose(Vector3r(x, y, z), AirSimClientBase.toQuaternion(toRadians(-Cam_Tilt), 0, toRadians(-Cam_Pan))), True) '

I'm using python 3.6 and numpy 1.13

sytelus, 2017-12-16:
I tried out call to `client.simSetPose` and it seems to work fine. I think the problem might be that you have updated your repo but not your PythonClient code. I would suggest to copy PythonClient folder from latest version and try it out.

closed 2017-12-18 by lo2aayy

koshyviv, 2018-02-05:
Hi, I'm facing the same issue as @lo2aayy. I did check and am using the latest code/repo.
Any suggestions?

zhangjunwang, 2019-04-16:
hi, I have the same problem with you. I copy PythonClient folder from latest version, but it does not work. Could you tell me how to solve it? thanks

`Traceback (most recent call last):
  File "pponew.py", line 325, in <module>
    client.setCarControls(car_controls)
  File "/home/zjw/.local/lib/python3.5/site-packages/airsim/client.py", line 305, in setCarControls
    self.client.call('setCarControls', controls, vehicle_name)
  File "/home/zjw/.local/lib/python3.5/site-packages/msgpackrpc/session.py", line 41, in call
    return self.send_request(method, args).get()
  File "/home/zjw/.local/lib/python3.5/site-packages/msgpackrpc/session.py", line 51, in send_request
    self._transport.send_message([message.REQUEST, msgid, method, args])
  File "/home/zjw/.local/lib/python3.5/site-packages/msgpackrpc/transport/tcp.py", line 94, in send_message
    sock.send_message(message, callback)
  File "/home/zjw/.local/lib/python3.5/site-packages/msgpackrpc/transport/tcp.py", line 19, in send_message
    self._stream.write(self._packer.pack(message), callback=callback)
  File "msgpack/_packer.pyx", line 284, in msgpack._packer.Packer.pack
  File "msgpack/_packer.pyx", line 290, in msgpack._packer.Packer.pack
  File "msgpack/_packer.pyx", line 287, in msgpack._packer.Packer.pack
  File "msgpack/_packer.pyx", line 263, in msgpack._packer.Packer._pack
  File "msgpack/_packer.pyx", line 263, in msgpack._packer.Packer._pack
  File "msgpack/_packer.pyx", line 234, in msgpack._packer.Packer._pack
  File "msgpack/_packer.pyx", line 277, in msgpack._packer.Packer._pack
  File "/home/zjw/.local/lib/python3.5/site-packages/msgpackrpc/transport/tcp.py", line 12, in <lambda>
    self._packer = msgpack.Packer(encoding=encodings[0], default=lambda x: x.to_msgpack())
AttributeError: 'numpy.float32' object has no attribute 'to_msgpack'
`