microsoft/AirSim #682: get back controls of the car

Jacquesval, 2017-12-13:
Hello,
I wanted to know if it's possible to get back the CarControls() variables (throttle, steering, brake) when the user is controlling the vehicle. I need them to make an imitation learning algorithm.

sytelus, 2017-12-19:
This is good feature. I've put this in backlog.

Jacquesval, 2018-01-30:
Hi,
I just wanted to know if you have implemented this option already ?
(By the way, what is backlog, I'm not very familiar with github ?)

Beijingsheng, 2018-08-20:
Has this problem been solved? I'm also curious about this function, it's truly useful for RL.

SoftArch, 2019-01-01:
Hi,
This feature is implemented?
Have any info when it will implement?

Gaduu, 2019-02-06:
Any updates on that?

msb336, 2019-10-03:
`simGetCarControls` is now implemented.

closed 2019-10-03 by msb336