microsoft/AirSim #685: record surface normals

j0lia, 2017-12-15:
just tried out recording multiple image types - very nice feature! unfortunately, ImageType 6 (surface normals) is not working as expected: it saves surface normals for the camera pointing to the ground, instead of straight ahead. Image types 0,3,5 are working as expected. Am I missing something? Thanks!

"Recording": {
		"RecordOnMove": true,
		"RecordInterval": 0.05,
		"Cameras": [
			{ "CameraID": 0, "ImageType": 0, "PixelsAsFloat": false, "Compress": true },
			{ "CameraID": 0, "ImageType": 3, "PixelsAsFloat": false, "Compress": true },
			{ "CameraID": 0, "ImageType": 5, "PixelsAsFloat": false, "Compress": true },
			{ "CameraID": 0, "ImageType": 6, "PixelsAsFloat": false, "Compress": true }
		]},

sytelus, 2017-12-16:
Thanks for reporting. I've committed and tested potential fix. Please get latest and try it out.

j0lia, 2017-12-18:
Cool, it works now!

closed 2017-12-18 by j0lia