microsoft/AirSim #686: Makefile:252: recipe for target 'CrashReportClient-Linux-Shipping' failed

AmbFlye, 2017-12-16:
We followed the unreal steps to download for linux and we keep getting this fail code ...Help Please!

sytelus, 2017-12-19:
Could you please list exact steps, OS and UE versions you are using. In order to help, we need much more information.

sytelus, 2018-01-13:
Closing due to inactivity.

mashrin, 2018-01-23:
Getting the same error.

OS - Ubuntu 17.10
Running Unreal 4.17

mashrin, 2018-01-23:
Fixed it.
This helped:

$ chown -R user: filepath //mention user and the path to the directory
$ su user
$ make

Do not use 'sudo' or 'root user' while executing make command as it leads to another error where it shows:
"Refusing to run with the root privileges."

kosmas12, 2019-03-24:
Oh my god! I was looking to compile it and couldn't because of that same error as AmbFlye. Thank you very much, mashrin. You saved me!!!!

PatrickUtz, 2019-07-17:
@mashrin I am wondering if you used "user" or if you replaced that with the actual username of the respective machine. Thank you in advance.

closed 2020-09-18 by jonyMarino

lahtis, 2021-05-04:
Getting the same error.

OS - Ubuntu 18.04.5
Running Unreal 4.27 and 4.26
[Log.txt](https://github.com/microsoft/AirSim/files/6423458/Log.txt)

chown -R tuomas ~/UnrealEngine-4.27 command not helped. Same error.