microsoft/AirSim #687: SimpleFlight License

auturgy, 2017-12-17:
SimpleFlight is a derivative of BaseFlight via HackFlight, as evidenced here: https://github.com/sytelus/HackflightTemp?files=1
and https://github.com/simondlevy/Hackflight/issues/2
Although effort to rewrite has been made, as a derivative it still needs to be GPL v3 licensed, unless all copyright holders approve the change (near impossible in this case, given the evolution of the code).

sytelus, 2017-12-17:
simple_flight is not derivative of hackflight. The repository you have referenced was to show Simon Levy (author of HackFlight) for possibility of creating library interfaces like simple_flight. I and Simon Levy are in contact with each other to see if HackFlight can use interfaces that are more similar to simple_flight. So it's not correct to say that simple_flight is rewrite or derivative of HackFlight. I wrote simple_flight from scratch, starting from empty files, because I wanted to make firmware as pure C++ library which didn't existed at the time.

auturgy, 2017-12-17:
No, you didn’t. 

https://github.com/Microsoft/AirSim/tree/master/AirLib/include/firmwares/simple_flight/firmware
can be easily traced to here:
https://github.com/sytelus/HackflightTemp/tree/master/include/firmware
then here:
https://github.com/simondlevy/hackflight
And from there to BaseFlight and multiwii.

The mixer code is a simple (albeit trivial) example of the lineage, as it's almost a direct copy all the way through.

Slightly more obfuscated is:
https://github.com/simondlevy/hackflight/blob/master/include/stabilize.hpp
https://github.com/Microsoft/AirSim/blob/master/AirLib/include/firmwares/simple_flight/firmware/PidController.hpp
https://github.com/trollcop/baseflight/blob/upstream/src/mw.c

But the derivation is clear.

Regards,

James

sytelus, 2017-12-19:
(BTW, some of the links you sent don't work)
If you pull the mixer files you had mentioned and look them side by side, it should be clear that  there is no real resemblance, let alone "direct copy all the way through". The only commonality from my perspective is use of a matrix variable with the same type name motorMixer_t and a comment around it. I should point out that my initial thought was to develop simple_flight and send it to Simon so he can see and use my ideas about new architecture and design. To make things easier for him I’d originally thought about having similar interface however I quickly realized that this was too limiting for the new architecture and design I had in mind and so I scrapped the whole idea of interface compatibility and just develop new code base without any such constraints of previous works. Code for two mixers is very different, for example, very important things like the matrix values as well as actual mixing arithmetic are quite different which is because simple_flight doesn’t even use the same coordinate system as Hackflight. You should also be able to easily see that simple_flight does min, max clipping in an obvious way, while Hackflight is doing it in more clever and efficient way. If things were copy, simple_flight would have certainly used better and more thoughtful way that is used by Hackflight! Also notice that simple_flight doesn’t take care of arm/disarm status like Hackflight does.

Some of the other files you have mentioned are again apple and oranges. Stabilize code in hackflight is doing some custom specialized stuff while simple_flight uses cascaded PID controllers with whole bunch of classes that don't even appear in any form or shape in Hackflight. You should also realize that simple_flight is not even a complete functional firmware like Hackflight because it cannot be deployed in hardware boards. The essential components like state estimator, sensor drivers, calibration routines etc are not yet developed in simple_flight. I’m inclined to think that perhaps that some of the standard terminologies or common labels being used are throwing you off? The labels and terminologies are common across such codebases and often used to represent known concepts and operations (also mentioned in many books, articles and code out there in existence on the subject.)

I also hope you do understand that this project is fully open source and our goal is to help advance the research.  Unfortunately, I don’t see your responses very respectful, well informed and certainly not in the spirit of the endeavor. We are very constrained on resources and I’d like to spend my time on more productive discussions and development than keep responding in this thread.

auturgy, 2017-12-19:
I have nothing to gain by this, and have neither the time nor inclination to provide a more detailed retort.

Hopefully I’ve sparked interest somewhere for a proper review.

Regards,

James

closed 2018-01-13 by sytelus