microsoft/AirSim #688: From velocity vector to yaw (along velocity)

Kjell-K, 2017-12-17:
@sytelus @ALL I took a look at your DQN Drone example and the action space of:

```
    quad_offset = interpret_action(action)
    quad_vel = client.getVelocity()
    client.moveByVelocity(quad_vel.x_val+quad_offset[0], quad_vel.y_val+quad_offset[1], quad_vel.z_val+quad_offset[2], 5)
```

How could you ensure that the drone is yawing in the direction of velocity automatically. This is motivated by visual based control and that the drone sees where it is going. 

I tried to think and solve with getOrientation, toEulerianAngle, drivetrain, yaw_mode. But nothing worked out for me. Some help would be great!

sytelus, 2017-12-19:
This example doesn't attempt to keep direction of travel same as direction of camera. The hope is that RL algorithm will eventually learn to move in the right direction :).

However, if you do want to move in the direction that camera looks at, you can use the parameter drivetrain = ForwardOnly in above method. This is the code that implements the logic: https://github.com/Microsoft/AirSim/blob/master/AirLib/src/vehicles/multirotor/controllers/DroneControllerBase.cpp#L609

Kjell-K, 2017-12-22:
Yes. But if have forward camera and want to do visual based RL, the agent has to receive states that are relevant to the learning. In my case I have to face the direction of travel. 

This does not work for me I want it to work. Lets say I am facing along global x and want to move only in global y direction:
`    client.moveByVelocity(0, 4, 0, 5, drivetrain = DrivetrainType.ForwardOnly ) 
`
I would like that the copter automatically rotates in along y direction now, but he is doing not. Instead he is still facing along x and just roll to y.

Is there any elegant way I can achieve what I want?

sytelus, 2017-12-29:
Above command doesn't work in your case because you are using default YawMode which requests continuous rotation (but at 0 deg/sec) i.e. rate mode. The forward-only mode doesn't get activated if YawMode is set to "rate". So changing this would make forward-only mode:

`client.moveByVelocity(-5, -5, -5, duration=20, drivetrain = DrivetrainType.ForwardOnly, yaw_mode=YawMode(False, 0))`

I think I'd code that showed warning when requesting forward-only mode but also specifying yaw mode with rate. For some reason this is not appearing so this is to be debugged.

This is however pretty raw feature. Any contributions to make this more solid are welcome :).

abdelrahman-t, 2017-12-31:
had a similar problem, did this instead:

vx, vy, _ = transformToEarthFrame([desired_velocity, 0, 0], client.getOrientation())
moveByVelocityZ(vx=vx, vy=vy, z=altitude, yawMode=YawMode(True, 0), drivetrain=DrivetrainType.MaxDegreeOfFreedom, duration=10.0)

abdelrahman-t, 2017-12-31:
from pyquaternion import Quaternion 
# a very handy quaternion libraray

def transformToEarthFrame(vector, q_):
    q = Quaternion(q_)
    return q.rotate(vector)

closed 2018-01-13 by sytelus