microsoft/AirSim #697: Upgrade to VS2017 + UE-4.18 coming soon!

sytelus, 2017-12-19:
Just so folks are prepared. We will be doling this upgrade around New Year.

saihv, 2017-12-24:
@sytelus Just wondering, was a fix found for https://github.com/Microsoft/AirSim/issues/433?

sytelus, 2017-12-29:
I haven't seen this warning again in 4.16.3. Have you seen it?

saihv, 2017-12-29:
Oops, I wasn't paying attention to the warning in 4.16 because it never really affected recording. I only remember it breaking recording in 4.17. I can check with 4.16 again

saihv, 2018-01-04:
I am still getting this warning in 4.16.3. 

`LogModuleManager:Warning: ModuleManager: Attempting to load 'ImageWrapper' outside the main thread.  This module was already loaded - so we didn't crash but this isn't safe.  Please call LoadModule on the main/game thread only.  You can use GetModule or GetModuleChecked instead, those are safe to call outside the game thread.`

molysgaard, 2018-03-06:
What is the status on UE4.18, I would like to use the CLionSourceCodeAccess plugin but it seems like it is not working on 4.17 that I am using now.

If there is anything I can do to help out, please tell me.

sytelus, 2018-04-28:
Upgrade is now done!

@saihv - you were correct about change needed in compress code. However it looked to me that compress should not really have a need to run on game thread. It should run on any thread! So I debugged Unreal code and found that cause was module loading that was being done for image wrapper component. So I pulled related code out to AirSim and modified it so it doesn't need module loading. Now it works on same thread!

Commit is here if you have comments: https://github.com/Microsoft/AirSim/commit/aaa7d80c32323a3ecbbb9f8dc4637542d58d4192

closed 2018-10-17 by sytelus