microsoft/AirSim #698: Is it possible to add 'noise' to the capture components?

saihv, 2017-12-19:
I am looking into adding some noise (Gaussian, speckle etc.) to the captured images from the onboard cameras. Because I want this to affect the accuracy of the depth map, disparity etc. I am trying to see if this can be added directly inside Unreal, rather than after accessing the rendered images. Would something like this possible, probably in the blueprints that take care of the image rendering?

sytelus, 2017-12-22:
Yes, this would be a great contribution! The way to do this is by adding a post process material that would add the noise. You can then add this material in to BP_PIPCamera. Here are some references:

[Getting the Most Out of Noise in UE4](https://www.unrealengine.com/en-US/blog/getting-the-most-out-of-noise-in-ue4)
[TV static effect](https://answers.unrealengine.com/questions/61940/tv-static-effect.html)
[Video -Material Noise](https://www.youtube.com/watch?v=VcZG0cXotqU)
[Commercial plugin that implements effect](http://chameleon.sumfx.net/effects/tv-noise)

saihv, 2017-12-24:
Thanks! I will look into it. Closing for now.

closed 2017-12-24 by saihv

sytelus, 2018-01-13:
I've added the noise feature! Here's the doc: https://github.com/Microsoft/AirSim/blob/master/docs/settings.md#image-noise-settings

saihv, 2018-02-13:
Thanks for the addition! Just a quick question: does this only work on the scene image? I am trying to examine the effects on disparity etc. by changing ImageType in the noise settings, but it doesn't seem to be working.

sytelus, 2018-02-17:
yes, it's only for scene image. The depth/disparity etc are ground truths and so we can't have noise in there.