microsoft/AirSim #699: AttributeError: 'CarState' object has no attribute 'collision'

kvasnyj, 2017-12-20:
Code from 'hello_car.py':
```
    # get state of the car
    car_state = client.getCarState()
    print("Speed %d, Gear %d" % (car_state.speed, car_state.gear))
```

Result:
```
Traceback (most recent call last):
  File "c:\SDC\AirSim\PythonClient\hello_car.py", line 11, in <module>
    car_state = client.getCarState()
  File "c:\SDC\AirSim\PythonClient\AirSimClient.py", line 556, in getCarState
    return CarState.from_msgpack(state_raw)
  File "c:\SDC\AirSim\PythonClient\AirSimClient.py", line 26, in from_msgpack
    obj.__dict__ = { k : (v if not isinstance(v, dict) else getattr(getattr(obj, k).__class__, "from_msgpack")(v)) for k, v in encoded.items()}
  File "c:\SDC\AirSim\PythonClient\AirSimClient.py", line 26, in <dictcomp>
    obj.__dict__ = { k : (v if not isinstance(v, dict) else getattr(getattr(obj, k).__class__, "from_msgpack")(v)) for k, v in encoded.items()}
AttributeError: 'CarState' object has no attribute 'collision'
```

sytelus, 2017-12-22:
I think this is because you are using new version of AirSim but old version of Python client. Just update Python client from your new version and should be fine.

Mikemraz, 2017-12-26:
@sytelus  Hi, sytelus. I am facing the same issue. How can I update Python client?

marooncn, 2017-12-27:
@sytelus , I have the same question. After I follow the tutorial [Build It on Linux](https://github.com/Microsoft/AirSim/blob/master/docs/build_linux.md) to install and configure the package, I run the demo [hello_drone.py](https://github.com/Microsoft/AirSim/blob/master/PythonClient/hello_car.py) and there are some errors:
/home/maroon/图片/屏幕截图 2017-12-27 18:11:42.png
What's more, in most cases, it shows like this:
/home/maroon/图片/屏幕截图 2017-12-27 18:14:26.png
What's the matter with it?

kvasnyj, 2017-12-27:
I use the new version of AirSim and the latest version of Python client from repo.
Problem has been resolved by adding two lines into `AirSimClient.py/LandedState/CarState`
```
collision = CollisionInfo()
kinematics_true = KinematicsState()
```
Thanks to https://github.com/Microsoft/AirSim/pull/696

Mikemraz, 2017-12-28:
@kvasnyj thank you! The problem was solved!

sytelus, 2017-12-29:
I found the issue and it's now fixed in [this commit](https://github.com/Microsoft/AirSim/commit/b1ecc13f500a05d355dbe7aca65c200a3f4e0051). Please get latest. If you had done local changes to workaround this issue and then you might want to revert them first. 

Also, please note that there is slight change in APIs. Specifically, CarState structure now has [full kinematics](https://github.com/Microsoft/AirSim/blob/b1ecc13f500a05d355dbe7aca65c200a3f4e0051/PythonClient/AirSimClient.py#L159). This now allows you to get much more ground truth information about car's state in `kinematics_true` than just position and velocity.

Duplicate issues:

- https://github.com/Microsoft/AirSim/issues/708
- https://github.com/Microsoft/AirSim/issues/699
- https://github.com/Microsoft/AirSim/issues/711

DQQ-077-SZ, 2018-01-11:
@sytelus  When I run the DQNcar.py, I got the ERROR " 'CarState' object has no attribute 'position' ". I guess it was caused by [this commit](https://github.com/Microsoft/AirSim/commit/b1ecc13f500a05d355dbe7aca65c200a3f4e0051).

sytelus, 2018-01-13:
DQNcar.py is also now fixed.

closed 2018-01-13 by sytelus

DQQ-077-SZ, 2018-01-13:
@sytelus I used the new version of DQNcar.py. However, I still got AttributeError: 'CarState' object has no attribute 'velocity'. I guess that it still be caused by the **recent change of CarState structure which now has full kinematics**.

superPershing, 2018-01-14:
@sytelus  @DQQ-077-SZ I also encounter the same issue.