microsoft/AirSim #701: Unreal exception while running AirSim

celloaerometer, 2017-12-21:
We've encountered with the exception attached as a photo below:
[image]

We are not sure it is caused by AirSim but until now it only happened after running with AirSim for several hours - happened several times in the past few days , we couldn't identify yet if there is a specific action that causes it. Have you encountered this problem before?

sytelus, 2017-12-22:
Could you please post call stack? Where the code is breaking? Also how you have been running AirSim? In recording mode or when calling APIs?

celloaerometer, 2017-12-24:
Sorry for the late response - it took a few days but the exception happened again. I couldn't figure out where the code is breaking. It is breaking while we are using AirSim with the API. Here is all the relevant data i could gather. If anything is missing please let me know.

[image]
[image]
[image]

celloaerometer, 2017-12-24:
Now it happens every time we leave the Unreal window. The last time it showed some more details so I upload two more photos:
[image]
[image]

It seems like the exception was generated by RenderRequest.cpp - I was stopped at line 134.

sytelus, 2017-12-29:
Could you please provide information on what exactly you are doing? Are you pressing recording button and then pressing ESC key? Are you running any Python script? What is the version of Unreal Engine are you using? Please provide as much information as you can so we can reproduce this issue.

celloaerometer, 2017-12-29:
We are running a modified python script based on the DQN_car.py you provided. The exception occurs when we leave the unreal window (without ending the session or the script) and do something in the background. While we are inside the Unreal window the exception never occurs. Our unreal version is 4.17.2.

sytelus, 2017-12-30:
When you say "leaving Unreal window", do you mean you switch to different window or that do you press ESC key? Does this happen consistently every time?

There is one known issue with recording functionality in 4.17 so I would suggest to use 4.16 for now.

Another thing to try is uncheck "Use less CPU in background" setting. See the [details here](https://forums.unrealengine.com/unreal-engine/feedback-for-epic/74186-processor-usage?p=1346492#post1346492).

celloaerometer, 2017-12-30:
I meant switch to a different window. It doesn't happen every time - it mostly happens after a several hours the Unreal was open and working with the python script. I will try your using 4.16 and update again later, thank you.

cangokalp, 2018-01-12:
[I'm on the 4.16 - and that setting is unchecked] I'm experiencing the same problem. For me the error occurs pretty much randomly - sometimes after 2 hours sometimes after 10 hours of training.

celloaerometer, 2018-01-13:
I've noticed the exception is thrown all the time in the RenderRequest.cpp so I wrapped the all thing in a try-catch block and it doesn't happen anymore. Unreal doesn't get stuck anymore either. It's a temporary patch that works well for me now - but I thought you should know, it looks like the source of our problem.

msb336, 2019-10-02:
This may have been caused by a race condition that was fixed in [this pr](https://github.com/microsoft/AirSim/pull/2146). Please re-open if you are still experiencing this issue.

closed 2019-10-02 by msb336