microsoft/AirSim #702: Segmentation view stays all green

aburgm, 2017-12-22:
I'm running AirSim in the simple blocks environment, and while the main image and depth screens produce correct results, the segmentation view simply stays all green. I also tried it with in the Mountain Landscapes environment, with the same result. This is on Mac OS X with Integrated Intel Graphics.

I verified that meshes have a "CustomDepth Stencil Value" set to different values for different meshes, and "Render CustomDepth Pass" is enabled, and I tried setting "custom depth stencil pass" to "Enabled with Stencil" in the Project Settings. Are there other things I can try to make this work?

kvasnyj, 2017-12-28:
Hi,
you need to setup different id for different objects. Example:
```
client.simSetSegmentationObjectID("[\w]*", 0, True);
client.simSetSegmentationObjectID("ground[\w]*", 1, True);
client.simSetSegmentationObjectID("road[\w]*", 2, True);
client.simSetSegmentationObjectID("sky[\w]*", 3, True);
client.simSetSegmentationObjectID("car[\w]*", 4, True);
client.simSetSegmentationObjectID("sign[\w]*", 5, True);
```
Source code of example for car https://github.com/kvasnyj/AirSim/blob/master/segmentation.py

sytelus, 2017-12-29:
AirSim uses feature in Unreal Engine called CustomStencil for segmentation view. I found couple of posts on Unreal forums that this feature is not working on Mac due to bug in Unreal:

- [SceneTexture:CustomStencil emits only zeroes on 4.18.2 Mac+Metal](https://answers.unrealengine.com/questions/740168/scenetexturecustomstencil-emits-only-zeroes-on-418.html)
- ["Buffer Visualization" > "Custom Stencil" does not animate properly, is not readable [on mac]](https://answers.unrealengine.com/questions/739638/buffer-visualization-custom-stencil-does-not-anima.html)

You might want to contact Unreal support if you have access to them or file new post on https://answers.unrealengine.com

aburgm, 2018-01-30:
Turns out this works as intended after updating to High Sierra, as someone else also reported [here](https://answers.unrealengine.com/questions/740168/scenetexturecustomstencil-emits-only-zeroes-on-418.html) as well.

closed 2018-02-13 by sytelus