microsoft/AirSim #704: Building Hexacopter error

NihalKarne, 2017-12-22:
I followed the instruction on this page https://github.com/Microsoft/AirSim/wiki/hexacopter
made the content of the file 1004_rc_hex_x.hil as instructed.
then on using make px4fmu-v2_default i get this error :
nihal@Dell:~/GitHub/Firmware$ make px4fmu-v2_default
[15/583] Creating airframes.xml
FAILED: cd /home/nihal/GitHub/Firmware/build/px4fmu-v2_default && /usr/bin/python /home/nihal/GitHub/Firmware/Tools/px_process_airframes.py -a /home/nihal/GitHub/Firmware/ROMFS/px4fmu_common/init.d --board CONFIG_ARCH_BOARD_px4fmu-v2 --xml
Aborting due to missing @class tag in file: '/home/nihal/GitHub/Firmware/ROMFS/px4fmu_common/init.d/1004_rc_hex_x.hil'
[15/583] Generating uORB topic sources
ninja: build stopped: subcommand failed.
make: *** [px4fmu-v2_default] Error 1

saihv, 2017-12-24:
Can you post your 1004_rc_hex_x.hil file? Looks like you're missing the tag "@class" in the file. If you look at the original quad_x file from which hex_x is derived, you should see tags such as @class, @maintainer etc.

NihalKarne, 2017-12-25:
#!nsh
#
# @name HIL Quadcopter +
#
# @type Simulation
#
# @maintainer Anton Babushkin <anton@px4.io>
#
sh /etc/init.d/rc.mc_defaults
set MIXER hexa_x
# Need to set all 8 channels
set PWM_OUT 12345678
set HIL yes

1004_rc_hex_x.hil file

NihalKarne, 2017-12-25:
`#!nsh
#
# @name HIL Quadcopter +
#
# @type Simulation
#
# @maintainer Anton Babushkin <anton@px4.io>
#

sh /etc/init.d/rc.mc_defaults

set MIXER hexa_x

# Need to set all 8 channels
set PWM_OUT 12345678

set HIL yes`

1004_rc_hex_x.hil file

NihalKarne, 2017-12-25:
```
#!nsh
#
# @name HIL Quadcopter +
#
# @type Simulation
#
# @maintainer Anton Babushkin <anton@px4.io>
#

sh /etc/init.d/rc.mc_defaults

set MIXER hexa_x

# Need to set all 8 channels
set PWM_OUT 12345678

set HIL yes
```
1004_rc_hex_x.hil file.

sytelus, 2017-12-29:
This might interest @lovettchris who originally built hexacopter.

NihalKarne, 2018-01-06:
For Ubuntu PX4AirframeFactMetaData.xml is located at : `.config/QGroundControl.org/PX4AirframeFactMetaData.xml`  
.config directory is by default hidden in Ubuntu.

That solves the issue. So closing this topic.
Thanks for help!

closed 2018-01-06 by NihalKarne