microsoft/AirSim #705: HITL with last versions of the PX4

avklab, 2017-12-24:
It may be useful information for community. The last versions of the PX4 firmware (1.7) wok correctly in HITL mode with simulators which don't send the ground truth MAVLink messages (mavlink_msg_hil_state_quaternion). Receiving hil_state_quaternion from AirSim makes the autopilot malfunctioning. In addition in my environment by some reason the AirSim sends incorrect accelerometer data in the mavlink_msg_hil_sensor. No gravity component in zacc when on the ground (see the screenshot from QGroundControl), so permanent 'Freefall/Landing detected' messages go from the controller. But HITL works fine in X-Plane (with my PX4 MAVLink plugin) through MavLinkTest and in jMAVSim.
[image]

sytelus, 2017-12-29:
AirSim doesn't use `mavlink_msg_hil_state_quaternion`, it uses [HIL_SENSOR](http://mavlink.org/messages/common#HIL_SENSOR) message.

AirSim uses [jMavSim ](https://github.com/PX4/jMAVSim) as official simulator implementation to test PX4 and so it uses same messages that jMavSim uses: (1) to [send sensor state](https://github.com/PX4/jMAVSim/blob/master/src/me/drton/jmavsim/MAVLinkHILSystem.java#L158), use HIL_SENSOR (2) to [get actuator controls](https://github.com/PX4/jMAVSim/blob/master/src/me/drton/jmavsim/MAVLinkHILSystem.java#L46) use either HIL_ACTUATOR_CONTROLS or HIL_CONTROLS.

You are correct about z acceleration where we subtract `g` from ground truth acceleration. This is again same as how [jMavSim has been doing](https://github.com/PX4/jMAVSim/blob/master/src/me/drton/jmavsim/SimpleSensors.java#L119) and how sensor drivers behave.

I'm not sure if something has changed from PX4 side but may be they didn't updated jMavSim? Or may be you are seeing something else? Please let us know if you find any other information that conflicts with above... It might be great to look in to plugin's code if its open source to see what PX4 really expects.

avklab, 2017-12-30:
O.K. with hil quaternion. So probably the only problem is incorrect accelerometer data. The physical accelerometer can not distinguish gravity and kinematic acceleration, they must be combined by the simulator. The jMavSim does this in [SimpleSensors.java, line 48](https://github.com/DrTon/jMAVSim/blob/c515008b8e3a275730378748a8d4899c65a1e9a9/src/me/drton/jmavsim/SimpleSensors.java#L48), so the zacc is approximately -9.8 on the ground (kinematic acceleration is zero, gravity is normal). I see that AirSim tries to do the same in [ImuSimple.hpp line 54](https://github.com/Microsoft/AirSim/blob/530644d29e630fa7cd8854026d5d1d53ee0ac3af/AirLib/include/sensors/imu/ImuSimple.hpp#L54) but without correct result (zacc is zero, i.e. weightlessness/free fall).

The last PX4 has corrected the bug in commander module which prevented arming in HITL mode. It also enables to switch any airframe into HITL mode by setting the SYS_HITL parameter in the QGroundControl.

I use the following code to send data from X-Plane to PX4 (the acceleration data gback, gright and gup from the X-Plane already include the gravity):
```
	/* Sensors */
	rotatemag(0.17f, 0, 0.49f, &xmag, &ymag, &zmag);
	xacc = -XPLMGetDataf(ggback)*ONE_G;
	yacc = XPLMGetDataf(ggright)*ONE_G;
	zacc = -XPLMGetDataf(ggup)*ONE_G;
	Ppx = XPLMGetDataf(gP);
	Qpx = XPLMGetDataf(gQ);
	Rpx = XPLMGetDataf(gR);
	baraltpx = XPLMGetDataf(gbaralt)*0.3048f;
	presspx = XPLMGetDataf(gpressure)*33.8639f;
	timeus = getTime_us();
	memset(buf, 0, BUFFER_LENGTH);
	mavlink_msg_hil_sensor_pack(1, 200, &msg, timeus, xacc + dist(eng)*1e-4, yacc + dist(eng)*1e-4, zacc + dist(eng)*1e-4,
		Ppx + dist(eng)*1e-5, Qpx + dist(eng)*1e-5, Rpx + dist(eng)*1e-5, xmag + dist(eng)*1e-4, ymag + dist(eng)*1e-4,
		zmag + dist(eng)*1e-4, presspx, 0, baraltpx + dist(eng)*1e-2, 20, 1 << 31);
	len = mavlink_msg_to_send_buffer(buf, &msg);
	link->send((char *)buf, len);
	/* GPS */
	latitudepx = XPLMGetDataf(glatitude)*1e7f;
	longitudepx = XPLMGetDataf(glongitude)*1e7f;
	altitudepx = XPLMGetDataf(galtitude) * 1000;
	groundsppx = XPLMGetDataf(ggroundspeed) * 100;
	vn = -XPLMGetDataf(gvz) * 100;
	ve = XPLMGetDataf(gvx) * 100;
	vd = -XPLMGetDataf(gVertV) * 100;
	if (timeus >= nextgps) {
		nextgps = timeus + 100000;
		memset(buf, 0, BUFFER_LENGTH);
		mavlink_msg_hil_gps_pack(1, 200, &msg, timeus, 3, latitudepx, longitudepx,
			altitudepx, 100, 100, groundsppx, vn, ve, vd, XPLMGetDataf(gcog) * 100, 10);
		len = mavlink_msg_to_send_buffer(buf, &msg);
		link->send((char *)buf, len);
	}
```

sytelus, 2018-01-06:
Thanks for more details. I'm planning to wrap up current work and look in to this around couple of weeks. Meanwhile if you want to dig in the code, you are welcome!

sytelus, 2018-02-13:
Looks like I won't be able to get to PX4 related issues for a while. If you can, please submit the pull request for any changes you have so far.

lovettchris, 2019-04-07:
I found a problem with PX4 1.8.2, they removed a module that is needed for HITL mode to work, namely, "pwm_sim_out" and so I built a new version of the v1.8.2 with this module included and takeoff and normally flying works again. See [OneDrive download](https://1drv.ms/f/s!AoS-ymEfLuCLqudxN4NCgYtf0f3WPQ).

closed 2020-09-04 by jonyMarino