microsoft/AirSim #707: Compile Error C1083

FrankXinqi, 2017-12-25:
I have found that I met the same error with someone, but the question was closed without a right solution.

I run 'build.cmd' in Developer Command Prompt for VS2015. (Community Version)

The error is shown as following.

DroneControllerBase.cpp
c:\git\airsim\airlib\include\common\vectormath.hpp(12): fatal error C1083: Cann
ot open include file: 'Eigen/Dense': No such file or directory [C:\Git\AirSim\A
irLib\AirLib.vcxproj]
MultirotorRpcLibServer.cpp
c:\git\airsim\airlib\include\common\vectormath.hpp(12): fatal error C1083: Cann
ot open include file: 'Eigen/Dense': No such file or directory [C:\Git\AirSim\A
irLib\AirLib.vcxproj]
MultirotorRpcLibClient.cpp
c:\git\airsim\airlib\include\common\vectormath.hpp(12): fatal error C1083: Cann
ot open include file: 'Eigen/Dense': No such file or directory [C:\Git\AirSim\A
irLib\AirLib.vcxproj]
CarRpcLibServer.cpp
c:\git\airsim\airlib\include\common\vectormath.hpp(12): fatal error C1083: Cann
ot open include file: 'Eigen/Dense': No such file or directory [C:\Git\AirSim\A
irLib\AirLib.vcxproj]
CarRpcLibClient.cpp
c:\git\airsim\airlib\include\common\vectormath.hpp(12): fatal error C1083: Cann
ot open include file: 'Eigen/Dense': No such file or directory [C:\Git\AirSim\A
irLib\AirLib.vcxproj]
SafetyEval.cpp
c:\git\airsim\airlib\include\common\vectormath.hpp(12): fatal error C1083: Cann
ot open include file: 'Eigen/Dense': No such file or directory [C:\Git\AirSim\A
irLib\AirLib.vcxproj]
ObstacleMap.cpp
c:\git\airsim\airlib\include\common\vectormath.hpp(12): fatal error C1083: Cann
ot open include file: 'Eigen/Dense': No such file or directory [C:\Git\AirSim\A
irLib\AirLib.vcxproj]
RpcLibServerBase.cpp
c:\git\airsim\airlib\include\common\vectormath.hpp(12): fatal error C1083: Cann
ot open include file: 'Eigen/Dense': No such file or directory [C:\Git\AirSim\A
irLib\AirLib.vcxproj]
RpcLibClientBase.cpp
c:\git\airsim\airlib\include\common\vectormath.hpp(12): fatal error C1083: Cann
ot open include file: 'Eigen/Dense': No such file or directory [C:\Git\AirSim\A
irLib\AirLib.vcxproj]

JcZou, 2017-12-25:
I met the same error, but now it is resolved.
You can manually download Eigen from http://eigen.tuxfamily.org/index.php?title=Main_Page#Download, then copy Eigen directory to the AirSim/AirLib/include

FrankXinqi, 2017-12-26:
Cool. Build successfully.
Thank you, JcZou.

closed 2017-12-26 by FrankXinqi