microsoft/AirSim #710: how to add a timer into AirSim

abc19850508, 2017-12-26:
I want to add a timer to control a capture image function. but I met this error 1>C:\Users\Windows\Documents\Unreal Projects\LandscapeMountains\Plugins\AirSim\Source\Recording\RecordingThread.cpp(208): error C3861: “GetWorldTimerManager”: 找不到标识符(means cannot find identifier)
1>C:\Users\Windows\Documents\Unreal Projects\LandscapeMountains\Plugins\AirSim\Source\Recording\RecordingThread.cpp(208): error C2228: “.SetTimer”的左边必须有类/结构/联合

I used timer in RecordingThread.cpp like this 
GetWorldTimerManager().SetTimer(this,
		&FRecordingThread::TCPConnectionListener, 0.01, true);

how should I solve the error "“GetWorldTimerManager”:  cannot find identifier"
thanks for your help!

sytelus, 2017-12-29:
I'm not sure what you are doing. If you want to set interval for recording then I suggest you use [RecordInterval setting](https://github.com/Microsoft/AirSim/blob/master/docs/settings.md#recording). If you are doing something more custom than try out Python APIs.

abc19850508, 2017-12-29:
@sytelus I want to use GetWorldTimerManager() to control network(send images to another computer) in RecordingThread.cpp.  But it always shows the error "GetWorldTimerManager() identifier not found". I think the problem is clear. Do I need to add some special head file or add "Networking" in the AirSim.Build.cs file. But I have tried both. There is still the error. So I ask here.

sytelus, 2018-01-13:
We do not have resources to assist in custom modifications related work, unfortunately. If you find any solution, please feel free to update this thread.

closed 2018-01-13 by sytelus