microsoft/AirSim #712: settings.json in document directory

mitchellspryn, 2017-12-28:
Currently, AirSim reads the settings.json from the "my documents" folder on windows, and the user's documents folder in Linux. This creates an issue when attempting to run AirSim in Azure under a service account that does not have access to these folders. For example, when running AirSim with BatchAI, the jobs are run under the _azurebatch service account that does not have write access to anywhere on the disk (so the settings.json cannot be generated automatically during the job, and needs to be manually populated on every machine in the cluster :( ). 

I have two proposals:

1) Allow the user to specify settings.json on the command line
2) First look in the user's documents folder. Then, if the settings.json is not found, look in the directory of the executable. Then throw error if not found in either of those places. 

Of these, I'd prefer #2. If @sytelus approves, I can work on implementation.

sytelus, 2017-12-29:
That's great point. I think we should do both of them. So the settings.json would be used in following order of priority:

1. Use one from command line if specified
2. Use one in executable folder if found
3. Use from document folder

Please let me know if you would like to go ahead and put this change in.

mitchellspryn, 2017-12-29:
Cool. I'll look at implementing this.

mitchellspryn, 2017-12-29:
[The PR is up](https://github.com/Microsoft/AirSim/pull/715).

mitchellspryn, 2017-12-30:
The PR is merged. Closing issue.

closed 2017-12-30 by mitchellspryn