microsoft/AirSim #714: CMake error

parang17, 2017-12-29:
When I run ./setup.sh, I got the following error message. I wonder how I can resolve this issue. 

------- Error message.
-- Found PythonInterp: /home/ychoi95/anaconda3/bin/python2.7 (found version "1.4") 
**_CMake Error at CMakeLists.txt:488 (message):
  Python 2.7 or newer is required_**

-- Configuring incomplete, errors occurred!
See also "/home/ychoi95/AirSim/llvm-build/CMakeFiles/CMakeOutput.log".
See also "/home/ychoi95/AirSim/llvm-build/CMakeFiles/CMakeError.log".

sytelus, 2017-12-29:
I think error message is pretty clear.

parang17, 2017-12-29:
I have python 3.6, but I do not know why it has the error. If you know how to resolve this issue, please let me know.

Thanks, 
YJ

mitchellspryn, 2018-01-01:
This might be a path issue. What version of python is on your path? [This SO thread might be relevant]. (https://stackoverflow.com/questions/15291500/i-have-2-versions-of-python-installed-but-cmake-is-using-older-version-how-do)

sytelus, 2018-01-13:
Closing this issue because of inactivity.

closed 2018-01-13 by sytelus