microsoft/AirSim #716: Build issue in Ubuntu

intijk, 2017-12-31:
https://github.com/Microsoft/AirSim/blob/master/docs/build_linux.md

I totally followed the link above, built the Unreal Engine out against 
https://github.com/EpicGames/UnrealEngine/commit/af96417313a908b20621a443175ba91683c238c8

And the AirSim out of 
https://github.com/Microsoft/AirSim/commit/530644d29e630fa7cd8854026d5d1d53ee0ac3af

But when I run it, it first told me that the Airsim plugin version is not compatible with Unreal and ask me if I want to try build, I built it and tried to run, and see below black scene.  Did I made some thing wrong?

Is there any updated version of Unreal version should I use?

[image]

intijk, 2017-12-31:
Or should I just try the Unreal 4.17 branch HEAD instead of the af96417 commit ?

intijk, 2017-12-31:
Tried HEAD of 4.17 branch of Unreal , still the same.

n1ckfg, 2018-01-05:
Try opening the .uproject file in a text editor and deleting the engine version, then choosing "convert in place" when prompted next time you open it. That's fixed it for me.

intijk, 2018-01-09:
Could you be more specific of the `.uproject` file?

I found below files:

```
./Unreal/Environments/Blocks/Blocks.uproject
./UnrealEngine/Engine/Documentation/Extras/API/Build/Documentation.uproject
./UnrealEngine/Engine/Source/Programs/IOS/UnrealLaunchDaemon/UnrealLaunchDaemon.uproject
./UnrealEngine/Samples/MobileStarterContent/MobileStarterContent.uproject
./UnrealEngine/Samples/StarterContent/StarterContent.uproject
./UnrealEngine/Templates/FP_FirstPerson/FP_FirstPerson.uproject
./UnrealEngine/Templates/FP_FirstPersonBP/FP_FirstPersonBP.uproject
./UnrealEngine/Templates/TP_2DSideScroller/TP_2DSideScroller.uproject
./UnrealEngine/Templates/TP_2DSideScrollerBP/TP_2DSideScrollerBP.uproject
./UnrealEngine/Templates/TP_FirstPerson/TP_FirstPerson.uproject
./UnrealEngine/Templates/TP_FirstPersonBP/TP_FirstPersonBP.uproject
./UnrealEngine/Templates/TP_Flying/TP_Flying.uproject
./UnrealEngine/Templates/TP_FlyingBP/TP_FlyingBP.uproject
./UnrealEngine/Templates/TP_Puzzle/TP_Puzzle.uproject
./UnrealEngine/Templates/TP_PuzzleBP/TP_PuzzleBP.uproject
./UnrealEngine/Templates/TP_Rolling/TP_Rolling.uproject
./UnrealEngine/Templates/TP_RollingBP/TP_RollingBP.uproject
./UnrealEngine/Templates/TP_SideScroller/TP_SideScroller.uproject
./UnrealEngine/Templates/TP_SideScrollerBP/TP_SideScrollerBP.uproject
./UnrealEngine/Templates/TP_ThirdPerson/TP_ThirdPerson.uproject
./UnrealEngine/Templates/TP_ThirdPersonBP/TP_ThirdPersonBP.uproject
./UnrealEngine/Templates/TP_TopDown/TP_TopDown.uproject
./UnrealEngine/Templates/TP_TopDownBP/TP_TopDownBP.uproject
./UnrealEngine/Templates/TP_TwinStick/TP_TwinStick.uproject
./UnrealEngine/Templates/TP_TwinStickBP/TP_TwinStickBP.uproject
./UnrealEngine/Templates/TP_Vehicle/TP_Vehicle.uproject
./UnrealEngine/Templates/TP_VehicleAdv/TP_VehicleAdv.uproject
./UnrealEngine/Templates/TP_VehicleAdvBP/TP_VehicleAdvBP.uproject
./UnrealEngine/Templates/TP_VehicleBP/TP_VehicleBP.uproject
./UnrealEngine/Templates/TP_VirtualRealityBP/TP_VirtualRealityBP.uproject
./UnrealEngine/UE4Games.uprojectdirs
```

sytelus, 2018-01-12:
Just did the clean build and I'm unable to reproduce your issue. I'm suspecting there might be your system specific issue, such as choosing correct drivers for your GPU.

closed 2018-02-13 by sytelus